BitBoxSwiss/bitbox02-firmware #55: [Feature request] Simulator

achow101, 2019-07-24:
A simulator similar to the one added to the Bitbox 1 would be nice for people integrating Bitbox 2 into their own software in order to do CI.

benma, 2019-07-29:
That sounds like a good thing to have. We are currently quite busy with preparing the launch, so unfortunately this needs to wait for a bit longer.

I sent you an email regarding getting you a device.

schmitmd, 2024-07-14:
I believe this has been completed in [PR #1167](https://github.com/BitBoxSwiss/bitbox02-firmware/pull/1167) and such code has continued to be maintained.

benma, 2024-07-14:
Right, and a PR in HWI to integrate the simulator into the HWI CI is also open and being worked on by @asi345: 

https://github.com/bitcoin-core/HWI/pull/741

Thanks again!

benma, 2024-07-14:
Also: the simulator is being released alongside the regular BitBox02 Multi releases, the first one in https://github.com/BitBoxSwiss/bitbox02-firmware/releases/tag/firmware%2Fv9.19.0.

It has also been integrated into the bitbox-api-rs repo for integration tests in https://github.com/BitBoxSwiss/bitbox-api-rs/pull/69 and in bitbox02-api-go for integration tests in https://github.com/BitBoxSwiss/bitbox02-api-go/pull/97.

closed 2024-07-14 by benma