BitBoxSwiss/bitbox02-firmware #68: [Feature request] Restore from seed

reardencode, 2019-08-05:
Reading through the code and README here it doesn't look like the bitbox02 supports restoring from a seed other than through the SD card. Would be nice to be able to round-trip a physical (written, engraved, stamped, etc.) seed.

Bonus points for checking (probably with a warning, not failure) the BIP39 checksum.

benma, 2019-08-06:
Hi @reardencode 

This feature is actually work in progress right now and will be released pretty soon. See https://github.com/digitalbitbox/bitbox02-firmware/pull/70. Same as exporting the mnemonic phrase. 

We will however enforce the checksum and support the standard English wordlist only.

reardencode, 2019-08-06:
BIP39 specifies that an invalid checksum should result in warning:

> Although using a mnemonic not generated by the algorithm described in "Generating the mnemonic" section is possible, this is not advised and software must compute a checksum for the mnemonic sentence using a wordlist and issue a warning if it is invalid. 

--Brandon, sent by an android

https://github.com/digitalbitbox/bitbox02-firmware/issues/68#issuecomment-518556354

benma, 2019-08-06:
Yes, we are aware. We chose to support only a subset of bip39 for the time being with the English wordlist and the checksum, as this allows to convert the phrase back to the original entropy.

reardencode, 2019-08-06:
Gotcha. Thanks! Can't wait to buy bitbox02! Looks amazing :)

--Brandon, sent by an android

https://github.com/digitalbitbox/bitbox02-firmware/issues/68#issuecomment-518609844

benma, 2019-08-06:
Thanks :heart:

benma, 2019-08-12:
Merged https://github.com/digitalbitbox/bitbox02-firmware/pull/70, with the release coming soon.

closed 2019-08-12 by benma