BitBoxSwiss/bitbox02-firmware #92: UI inconsistency when entering password

x1ddos, 2019-08-18:
Upgraded to firmware v3.0.0 (beta device) and tried the new input method like in https://twitter.com/ShiftCryptoHQ/status/1162301973347147776. I can't always reproduce it reliably but here it is:

1. Touch drag a finger a little somewhere between the top middle and top right where the arrow down is. At the same time, also touch at the bottom right as if you were confirming the password. Character set disappears:

[image]

Note that my fingers aren't touching the device at all in the photo anymore.

2. Touch again a couple times somewhere in the middle at the top. Everything reappears but the character set indicator at the top does not match the actual characters displayed at the bottom:

[image]
 
Feels like a race condition somewhere in the UI code.

thisconnect, 2019-09-26:
I can reproduce this with 4.1.0 tested with 4.1.0 and can reproduce this:

> Touch drag a finger a little somewhere between the top middle and top right where the arrow down is.

[image]

x1ddos, 2019-09-26:
> I can reproduce this

Thanks @thisconnect! I thought I was going crazy for a second. :)

benma, 2019-10-07:
This PR probably fixes the inconsistency: https://github.com/digitalbitbox/bitbox02-firmware/pull/153

@thisconnect your gif shows a differnent issue though, which is that the touch release event was not reliably fired. Could you pls open a new issue for this (or should I?). Then we could close this one.

thisconnect, 2019-10-07:
@benma sure, here you go #155

benma, 2019-10-07:
Fixed, please re-open if the issue re-appears.

closed 2019-10-07 by benma