BitBoxSwiss/bitbox02-firmware #101: No bitcoin transaction signing python code

achow101, 2019-08-21:
I noticed that there was an [`eth_sign`](https://github.com/digitalbitbox/bitbox02-firmware/blob/master/py/bitbox02/bitbox02.py#L491) in the provided python library, but no `btc_sign`. Could a `btc_sign` function be provided?

benma, 2019-08-21:
For sure. It was supposed to be added latest when we start writing an Electrum plug-in, but we could accelerate this a bit.

Is there a Python library you can recommend for PSBT? Could start experimenting with this too.

achow101, 2019-08-22:
> Is there a Python library you can recommend for PSBT? Could start experimenting with this too.

There aren't any AFAIK. Although now that you mention it, I probably should publish such a library. So far, I've pretty much just been copying around https://github.com/bitcoin-core/HWI/blob/master/hwilib/serializations.py to wherever I need PSBT in python, which isn't all that often.

benma, 2019-08-30:
@achow101 could you please check if the initial draft of the btc_sign method works for you?

https://github.com/digitalbitbox/bitbox02-firmware/pull/103

Any input on how to improve the API?

Thanks.

benma, 2019-09-11:
Merged the PR, we can iterate on the Python API in new PRs if needed. Please let me know if it's working for you.

benma, 2019-09-19:
Closing this as basic support is merged. Please open new issues if something comes up. Thanks!

closed 2019-09-19 by benma