BitBoxSwiss/bitbox02-firmware #113: [Feature request] SLIP-39 Shamir Backup support?

jonathancross, 2019-09-08:
Trezor now ships with support for [SLIP-0039 : Shamir's Secret-Sharing for Mnemonic Codes](https://github.com/satoshilabs/slips/blob/master/slip-0039.md).  It would be great if the BitBox02 would also add support for this emerging standard to prevent vendor lock-in and give users an easy way to restore their wallet on either device.

benma, 2019-09-19:
Sorry for the late reply.

It's a very interesting feature for sure! Currently we have our hands full, so we are not looking into adding this in the near future.

It's definitely on our radar though.

Thanks!

nioncode, 2020-11-08:
@benma would you mind re-opening this issue so that we can subscribe to it and be notified of the feature being implemented (hopefully) in the future?

SLIP-39 seems to be a nice way to backup the device as part of a 2-of-3 scheme to not expose full access to the funds when someone gets hold of only one of the seed backups (which is the case right now, since there is only a backup of the whole seed in plain sight).

benma, 2020-11-08:
Ok, done ;) 

To manage expectations: this feature is still not on our roadmap, and it is not likely it will be be on the roadmap in the short/mid term. 

Other solutions that can protect your backups in the meantime, for **expert** users only: 

- Optional passphrase
- [Multisig[(https://shiftcrypto.ch/blog/bitbox02-electrum-bitcoin-multisig/)

My1, 2020-11-30:
while I am not a particular fan of the SLIP39 implementation of shamir (mainly due to lack in compatibility to BIP39) I think shamir is a pretty helpful way of having **secure but redundant** backups, especially since Multisig needs multiple devices and stuff it's in my opinion more suited when multiple parties are supposed to have control over any given set of funds, like for example when you have a company like a financial board in a company with 5 people of which 3 are needed to authorize stuff.

like with Passphrase you basically have a 2-of-2 scheme which is surely nice for security but no redundancy except by storing your seed and passphrase multiple times.

ThomasSRL, 2021-12-10:
I would love to see thus implemented, since it would allow a more redundant backup for my single hot wallet.

Cold wallet should be multisig anyway IMO.

benma, 2022-01-04:
I think the effort is better spent in improving the multisig experience, as multisig not only provides a distributed backup, but also distributed signing.

benma, 2022-02-09:
[Copying here for the record](https://www.reddit.com/r/BitBoxWallet/comments/rti44u/slip39/hw96pm5/?context=3)

> Multisig in altcoins is non existent or relies on smart contracts. Also, some institutions use a combination of multisig and Shamir shares for added redundancy. Shamir would be great to have in addition to multisig when developer resources allow.

Senjuu, 2024-12-27:
Hallo and sorry if I am at the wrong place for writing this but currently I deliberate to switch my hardwarewallet to a BitBox02. The only dealbreaker currently holding me up is the missing support for SLIP-39.

My thoughts regarding this discussion are
> I think the effort is better spent in improving the multisig experience, as multisig not only provides a distributed backup, but also distributed signing.

That there are several standards by the BitBox are not compatible with multisig (e.g. BIP-85 which "can be used to onboard friends and families" and as described by @benma thoses people (which are no experts) should not be introduced to multisig. Therefore my single seed must be better secured.)

> Other solutions that can protect your backups in the meantime, for **expert** users only:
> 
>* Optional passphrase
> 
> * [Multisig](https://shiftcrypto.ch/blog/bitbox02-electrum-bitcoin-multisig/)

In my opinion there are some problems to your suggestion @benma.
1st more complex multisig (except by using keyaggregation through taproot-keypath) consumes more blockspace than a single key. Which can be argued (as by @My1) can be unpractical for a single person scenario.
2nd there is a feasable limit for non keyaggregation multisig which is, as far as I know, already surpassed by SLIP-39 (e.g. I have two shares from which I can recover my wallet alone or if one of those is lost I can request 3 of my 5 friends' share (which I gave them) to recover my wallet.
3rd if I lost some of my SLIP-39 shares or want to rearrange my distribution I can let my hardwarewallet create a new set of shares and continue as if nothing happend (which has some caviats I will add) while in a multisig setup I need to make some onechain transactions to heal or change my distributed backup

The caviats to my 3rd point is that a new set of shares does not invalidate the old shares so someone creating a new set should either have both sets be part of the backup/distribution strategy or should make sure that there is no overlap in shares, which should be regarded as compromised, which could result in a third party gaining the ability to restore the seed. This caviat is one of the reasons for SLIP-39 being incompatible to BIP-39 as @My1 already pointed out as a single surving compromised BIP-39 backup would be a single point of failure a proceeding SLIP-39 setup.

Finally I would like to help implementing SLIP-39 into the BitBox02 as I am confident in my abilities to write in C. But as this would be my first contribution to anything part of a hardwarewallet I am hesitent as to not introduce some vulnerabilities. Therefore I would like to know if my help could be used and if so I would limit myself to creating kind of a SLIP-39-library-code to your specifications for others to just need to integrate this code to the data-storage and BitBoxApp-UI.

jonathancross, 2024-12-28:
Just to be clear, I no longer believe SLIP-39 is a protocol that should be adopted.  After experimenting and learning about the limitations, I believe there are critical features missing.  Very sad that Satoshi Labs went ahead with this particular implementation of SSS.  Blockchain Commons has a superior protocol which allows bi-directional conversion between BIP-39 seed phrase and SSS shares. https://github.com/BlockchainCommons/lethekit/issues/38#issuecomment-640919432

closed 2024-12-28 by jonathancross

benma, 2024-12-29:
@jonathancross in the linked comment the rationale is

> My deal breaker here is that losing the ability to round trip from BIP39<->seed<->shards means that wallets are locked in to supporting SLIP39 the way Satoshi Labs does it. With our current sharding method (which will need to be changed to not confuse with SLIP39), you are not locked in to the Trezor

I find this unconvincing. There is no vendor lock-in if others adopt SLIP-39, in the same way others would have to adopt SSKR. Also SLIP-39 is open source (protocol and code), so one can always create&recover on an offline computer, same as with SSKR (which afaik currently does not have any HWW support).

Senjuu, 2024-12-29:
Copying here for the record (just in case the #1351 will be handled here)
> Lastly I want to ask for some additional quality of life features if this would be included in the BitBox:
> 1. Continue Recovery in case of powerloss (not a partial typed in shard just from the last complete shard)
> 2. Continue a begun recovery without a PC (e.g. just power from a powerbank) to not need to bring all shards together (just go from shard to shard and type them in / read in the SD-card)
> 3. Continue a "completed" Dry-Run Recovery for the remainding shards to check the whole setup at once

jonathancross, 2025-01-05:
Sorry for the delay here.

> I find this unconvincing. There is no vendor lock-in if others adopt SLIP-39, in the same way others would have to adopt SSKR. Also SLIP-39 is open source (protocol and code), so one can always create&recover on an offline computer, same as with SSKR ...

I was not talking about vendor lock-in, but rather that I agree with @ChristopherA that this _missing feature_ is a deal breaker in terms of recommending SLIP-39.  SSKR is a superior standard which accomplishes the same goals IMHO.  SSKR allows an existing BIP-39 seed phrase to be _upgraded_ to a sharded backup without moving funds, etc.

I am therefore withdrawing my support for this SLIP-39 feature request and instead advocating for SSKR to be implemented (here and in other HWW).

> SSKR (which afaik currently does not have any HWW support).

I would love to see BitBox lead the way in being the first _commercial_ HWW to implement SSKR.

It is already implemented in the DIY [Lethekit](https://github.com/BlockchainCommons/lethekit?tab=readme-ov-file#blockchain-commons-lethekit) HWW.

ChristopherA, 2025-01-05:
Note that Ledger now officially offers an alternative to Ledger Recover with an SSKR (Sharded Secret Key Reconstruction) app. It supports user's to be the sole keepers of their seed phrase, but splits into shares for reliability.

https://www.ledger.com/blog/seed-tool-app

Details on the base format and libraries supporting it are at https://developer.blockchaincommons.com/sskr/

We also have new format leveraging SSKR that also shard descriptors and other important metadata, which is particularly important for recovery in multisig scenarios, and a new protocol on top that allows for secure requests and responses, making it easier for users as they can help automate recovery.

— Christopher Allen