BitBoxSwiss/bitbox02-firmware #151: Feature request: add extended ascii caracters for password input

wilya7, 2019-10-04:
Hello, I wanted to use bitbox02 as a second device to use in case of emergency with the same BIP39 seeds of an other wallet, however since I have a 25th word containing an ascii extended character (such as !@#$%^&*(){}:;"'.,?/\ ..) I could not do that. Would be possible to add these characters to the firmware?

benma, 2019-10-07:
Hi @wilya7 

Thanks for the request. This is definitely a useful feature. I'm working on adding it.

benma, 2019-10-08:
Support added in https://github.com/digitalbitbox/bitbox02-firmware/pull/154. This will be available with the next firmware update.

The initial set of supported special characters will be the space ` ` plus `!"#$%&'()*+,-./:;<=>?^[\]@_{|}`, which is nearly all of them.

I hope this helps. Please re-open the issue if needed.

closed 2019-10-08 by benma

benma, 2019-10-29:
Fyi, the feature is now released in firmware v4.2.1, which can be installed with the [BitBoxApp](https://shiftcrypto.ch/start/).