BitBoxSwiss/bitbox02-firmware #155: UI Character set disappears when wild touching

thisconnect, 2019-10-07:
related #92 ?
[image]

NickeZ, 2020-01-07:
Can you still reproduce this on the latest firmware? I believe I have a fix, but I cannot reproduce the problem anymore.

thisconnect, 2020-01-07:
@NickeZ you are right, it is much harder almost impossible now to reproduce as I described.

I did manage doing it when touching on the top middle/right first followed by bottom middle/right.

Updated step to reproduce:
touching the ABC (topmiddle) that rotates the ABC, then immediately touching top/right will reproduce it again. 

FYI I can't see the animation of the ABC rotation but I strongly believe the bug only works if it rotated. Maybe touching the bottom was never important

BTW: the UI still works, 
i.e. the delete touch still works.

this is Firmware 5.0.0 and App 4.16.0

❤️ for looking into this. and as said it is already way better!

thisconnect, 2020-01-07:
let me know if you understood my description, I could phrase it differently if you want or try to record another video.

thisconnect, 2020-01-07:
As it is still pretty hard to reproduce it is also fine by me if you wontfix this, feel free to close or lowprio

NickeZ, 2020-01-08:
> touching the ABC (topmiddle) that rotates the ABC, then immediately touching top/right will reproduce it again.

I tried to do this for several minutes and couldn't reproduce it. Maybe you could shoot a video of it? It sounds like we have a race condition that is going to be quite hard to solve unless I can reproduce it.

NickeZ, 2020-02-18:
This was fixed in https://github.com/digitalbitbox/bitbox02-firmware/pull/278 and is part of the 6.0.0 firmware release. Thank you for your patience!

closed 2020-02-18 by NickeZ