BitBoxSwiss/bitbox02-firmware #313: Bitcoin Cash support

StEvUgnIn, 2020-01-31:
Hi, 
Do you consider supporting Bitcoin Cash and other bitcoin forks?

Thanks a lot,
Steve

benma, 2020-02-07:
We are continuously considering supporting more coins. Bitcoin Cash and other forks are currently not on the roadmap, though this might change in time.

closed 2020-02-07 by benma

gasull, 2020-11-10:
Has this changed?

Your website says "We focus our efforts on the top tier coins under active development".

Bitcoin Cash is number 5 in market capitalization and is under active development.  You're supporting other coins with much lower market value.

benma, 2020-11-10:
Unfortunately no change regarding Bitcoin Cash at this moment. I'll be sure to re-open this issue if this changes.