BitBoxSwiss/bitbox02-firmware #417: [feature request] Restore without app

SCBuergel, 2020-03-22:
In contrast to competitors, the Bitbox2 needs actions on the app in order to restore a wallet. I do not feel good about having my laptop around while entering my mnemonic into the Bitbox2 - tell me that's ridiculous but that's how I feel :-/

Instead, I'd be happy if I could just connect my Bitbox2 to whatever USB-C power-supply (that I trust more than my laptop) and initiate the restore process just through the Bitbox interface without needing the app.

Suggested UI (on Bitbox2 screen) - if Bitbox2 is in factory reset mode:
```
1. create wallet
2. restore mnemonic
3. restore SD
(or see app on computer)
```
These 3 options could be selected by tapping on three corners of the screen or by sliding through a dropdown-like list. In each case, the same process that is currently being used when selecting the corresponding action on the app could be triggered here. Option 3 might be a bit more tricky as the different backups from the SD card need to be rendered - but I'm mostly interested in Option 2 anyway ;-)

benma, 2020-03-22:
Hi. Thanks for the request. 

Air-gapped operation of the BitBox02 would be desirable. 

I will add this feature to our potential features list. I will get back to you once we added a score for its priority. 

I guess verifying the backup/mnemonic would also be good to have then.

Side note: point 1 would suffer a bit because entropy from the host wallet is sent to the device to be mixed into the seed generation, but I guess this can be skipped in this case, possibly with a small warning.

nioncode, 2020-08-20:
The additional entropy is only needed for 1) right, not for 2) and 3) since the mnemonic already contains all the information and no additional entropy is required. So at least 2) and 3) could be easily done.

For 1), it would be great if we could add additional entropy, e.g. from dice rolls / shuffling a card deck or by reading entropy from another computer (e.g. from https://iancoleman.io/bip39/). I think the touch controls would provide a nice way of entering 0s and 1s in a fast way.

NickeZ, 2025-10-23:
This is not on our roadmap.

closed 2025-10-23 by NickeZ