BitBoxSwiss/bitbox02-firmware #418: [UX improvement] Allow editing of previous mnemonic words

SCBuergel, 2020-03-22:
When restoring a wallet from mnemonic, I currently have only one attempt for each word. If I enter the wrong word, the entire process has to be aborted and repeated which is a painful process, especially for 24word mnemonics.

I suggest to allow users to review and edit previously entered words.

UX suggestion on Bitbox 2 screen:
Currently, after completing a word, there is a `back` button in the top left corner of the screen:
```
<
      Enter 6th word
abcd    ijkl     qrstu
efgh    mnop     vwxyz
```

That button takes us to the `Do you really want to cancel` screen. I suggest to replace that screen and instead show:
1. the previous word
2. arrow left and right to navigate to the previous and next words
3. `replace word` on the bottom

```
<               >
      5th word:
       zoo
replace word
```

Then currently existing `Do you really want to cancel` screen could be shown when tapping again on the `<` button (top left) once the first word has been reached.

benma, 2020-03-24:
I like the idea, but the downside with the suggested tx is that if you wanted to abort at the 18th word, it would involve lots of tapping.

Maybe the back button could open a very small menu with the choices of cancelling and deleting one word.

My1, 2020-09-16:
good idea. I just noticed at like word 13 that I skipped something when trying and a full restart would be annoying.
one could make for example to enter a "review mode" via the top left which shows the words entered so far and select which one to change, a bit similar to the seed backup verification scene where you have to select a word out of the list.

benma, 2020-09-25:
Finally some progress here:

https://github.com/digitalbitbox/bitbox02-firmware/pull/588

This implements the feature as described in my previous comment: one can go back an edit the previous word (repeatedly) and use this to correct mistakes.

My1, 2020-09-25:
sounds awesome

benma, 2020-09-28:
Merged, this will be available with the next firmware release. Thanks all.

closed 2020-09-28 by benma