BitBoxSwiss/bitbox02-firmware #453: Return a failure code if initialized and locked

x1ddos, 2020-04-08:
This came up in https://github.com/digitalbitbox/bitbox02-firmware/pull/451:

```
    // No other message than the attestation and unlock calls shall pass until the device is
    // unlocked or ready to be initialized.
    if bitbox02::memory::is_initialized() && bitbox02::keystore::is_locked() {
        return Vec::new();
    }
```

Instead of returning an empty slice, `OP_STATUS_FAILURE` or something else could be returned to make it more consistent?

va-an, 2025-10-23:
@NickeZ I apologize for the trouble, I couldn't find information about a Discord or other communication platform, so I don't know who to contact.

I was looking for a good-first-issue for myself.
I couldn't find the file or code mentioned in this issue - the issue appears to be outdated - I suggest closing it.

Currently, all issues with the good-first-issue tag in the repo were created in 2020, and I suspect the others are outdated as well.
Is there any current issue that would be suitable for getting familiar with the code?

NickeZ, 2025-10-23:
You will be able to do all development with the simulator. So a dev-device shouldn't be necessary. If you want to work on something that aligns with our plans and you want a dev device I'm sure we can arrange something. Please reach out to support then.