BitBoxSwiss/bitbox02-firmware #473: BitBox02 working with Electrum Wallet

csh2000, 2020-04-25:
Hello, Electrum Wallet does not recognize BitBox02.

It is working fine with other hardware wallets.

Not sure what "udev rules" are needed for BB02.

https://electrum.readthedocs.io/en/latest/hardware-linux.html

Thanks.

x1ddos, 2020-04-25:
Re: udev rules - https://shiftcrypto.ch/start/ and click on Linux => Linux64 Other Distros.

It depends on your linux distribution. On the website, `plugdev` group is used but your distribution might be using a different group. For instance, you could just use `users` group on VoidLinux.

Please post here if the above worked for you.

TheCharlatan, 2020-04-26:
Docs added in https://github.com/spesmilo/electrum-docs/pull/104/files

csh2000, 2020-04-28:
Hello, tried all of the above but Electrum Wallet does no detect BB02 ("no hardware device detected").

Also tried added my user to "plugdev" group.

I am running Ubuntu 18.04 and Electrum Wallet (running AppImage) does detect oher hardware wallets.

BB02 is working fine with BBApp, the problem is with Electrum Wallet.

Cau you provide the text to be included in "rules" files at "etc/udev/rules.d" folder (51-hid-digitalbitbox.rules / 52-hid-digitalbitbox.rules / 53-hid-bitbox02.rules / 54-hid-bitbox02.rules)?

Thanks!

x1ddos, 2020-04-28:
If the BB app detects the device, there's no need to update udev configuration.

Could you post the Electrum version you're using and, if possible the bitbox02 python package? Not sure if Electrum bundles everything together or it's installed separately. Maybe @TheCharlatan or @benma know.

Stadicus, 2020-05-26:
Electrum supports the BitBox02 now, but there has not been a new release for a long time. If you are running the AppImage of Electrum, the latest version is 3.3.8 from July 2019. This was even before the launch of the BitBox02.

We added support here https://github.com/spesmilo/electrum/pull/5993, and this pull-request has already been merged into master. If you run Electrum from source code, the BitBox02 is supported. We are hoping for a new official release soon as well.

csh2000, 2020-06-02:
Thanks @Stadicus, I guess that is the issue, I am having some problems running from source code, so I am running the AppImage.

I close this Issue.

closed 2020-06-02 by csh2000