BitBoxSwiss/bitbox02-firmware #494: Docker: Error building container

TheCharlatan, 2020-06-03:
I wanted to build the docker container to reproduce today's release build, but it failed at installing protoc-gen-doc:

```
Removing intermediate container 1212580db3af
The command '/bin/sh -c go get -v -u github.com/pseudomuto/protoc-gen-doc/cmd/protoc-gen-doc' returned a non-zero code: 2
```

I investigated further and ran:
```
docker exec -ti bitbox02-firmware-dev go get -v -u github.com/pseudomuto/protoc-gen-doc/cmd/protoc-gen-doc
```

Which in turn showed me the actual error:
```
opt/go/src/google.golang.org/protobuf/proto/proto_methods.go:18:23: cannot use m.ProtoMethods() (type *protoreflect.Message) as type *struct { pragma.NoUnkeyedLiterals; Flags uint64; Size func(struct { pragma.NoUnkeyedLiterals; Message protoreflect.Message; Flags uint8 }) struct { pragma.NoUnkeyedLiterals; Size int }; Marshal func(struct { pragma.NoUnkeyedLiterals; Message protoreflect.Message; Buf []byte; Flags uint8 }) (struct { pragma.NoUnkeyedLiterals; Buf []byte }, error); Unmarshal func(struct { pragma.NoUnkeyedLiterals; Message protoreflect.Message; Buf []byte; Flags uint8; Resolver interface { FindExtensionByName(protoreflect.FullName) (protoreflect.ExtensionType, error); FindExtensionByNumber(protoreflect.FullName, protowire.Number) (protoreflect.ExtensionType, error) } }) (struct { pragma.NoUnkeyedLiterals; Flags uint8 }, error); Merge func(struct { pragma.NoUnkeyedLiterals; Source protoreflect.Message; Destination protoreflect.Message }) struct { pragma.NoUnkeyedLiterals; Flags uint8 }; CheckInitialized func(struct { pragma.NoUnkeyedLiterals; Message protoreflect.Message }) (struct { pragma.NoUnkeyedLiterals }, error) } in return argument
```

After reading through this: https://github.com/golang/protobuf/issues/1094 I am thinking that the problem is the go version. Should the Dockerfile be updated to 1.13?

TheCharlatan, 2020-06-04:
Seems like this kind of problem will pop up in the future again the way the dependencies are imported right now. Everything that is downloaded should be versioned and checked. I'm also not a fan of using rustup to get the rust toolchain. Can this be replaced by a static script?

benma, 2020-06-05:
Fixed in https://github.com/digitalbitbox/bitbox02-firmware/pull/495

NickeZ, 2020-06-09:
Actually fixed in https://github.com/digitalbitbox/bitbox02-firmware/pull/496

closed 2020-06-19 by benma