BitBoxSwiss/bitbox02-firmware #514: latest firmware not compatible with MyEtherWallet

SCBuergel, 2020-06-24:
i was just prompted to update the bitbbox02 firmware and did so. However, now MyEtherWallet cannot connect to the bitbox anymore saying that `MyEtherWallet does not yet support this version of the firmware`.

This is fairly critical as MEW seems to be currently the only option to use the bitbox2 with Ethereum (other than sending simple ETH txs through the bitbox app which is not sufficient for most cases such as interfacing multisig wallets). 

The error details from MEW below:

```
{
  "exception": {
    "values": [
      {
        "type": "Error",
        "value": "Device or websocket not connected",
        "stacktrace": {
          "frames": [
            {
              "colno": 33691,
              "filename": "https://www.myetherwallet.com/js/vendors.307df816.js",
              "function": "WebSocket.s.socket.onmessage",
              "in_app": true,
              "lineno": 57
            },
            {
              "colno": 81175,
              "filename": "https://www.myetherwallet.com/js/vendors.307df816.js",
              "function": "Object.e.$externalizeWrapper.e.$externalizeWrapper [as OnRead]",
              "in_app": true,
              "lineno": 161
            },
            {
              "colno": 87974,
              "filename": "https://www.myetherwallet.com/js/vendors.307df816.js",
              "function": "?",
              "in_app": true,
              "lineno": 161
            },
            {
              "colno": 81175,
              "filename": "https://www.myetherwallet.com/js/vendors.307df816.js",
              "function": "Object.e.$externalizeWrapper.e.$externalizeWrapper",
              "in_app": true,
              "lineno": 161
            },
            {
              "colno": 3831439,
              "filename": "https://www.myetherwallet.com/js/vendors.307df816.js",
              "function": "Object.D.ptr.OnRead",
              "in_app": true,
              "lineno": 161
            },
            {
              "colno": 77880,
              "filename": "https://www.myetherwallet.com/js/vendors.307df816.js",
              "function": "Lt",
              "in_app": true,
              "lineno": 161
            },
            {
              "colno": 78153,
              "filename": "https://www.myetherwallet.com/js/vendors.307df816.js",
              "function": "?",
              "in_app": true,
              "lineno": 161
            },
            {
              "colno": 77411,
              "filename": "https://www.myetherwallet.com/js/vendors.307df816.js",
              "function": "Rt",
              "in_app": true,
              "lineno": 161
            },
            {
              "colno": 77303,
              "filename": "https://www.myetherwallet.com/js/vendors.307df816.js",
              "function": "Ot",
              "in_app": true,
              "lineno": 161
            },
            {
              "colno": 76879,
              "filename": "https://www.myetherwallet.com/js/vendors.307df816.js",
              "function": "r",
              "in_app": true,
              "lineno": 161
            },
            {
              "colno": 76946,
              "filename": "https://www.myetherwallet.com/js/vendors.307df816.js",
              "function": "e",
              "in_app": true,
              "lineno": 161
            },
            {
              "colno": 3831927,
              "filename": "https://www.myetherwallet.com/js/vendors.307df816.js",
              "function": "Object.n [as $blk]",
              "in_app": true,
              "lineno": 161
            },
            {
              "colno": 3768713,
              "filename": "https://www.myetherwallet.com/js/vendors.307df816.js",
              "function": "Object.U.ptr.Init [as $blk]",
              "in_app": true,
              "lineno": 161
            },
            {
              "colno": 3770110,
              "filename": "https://www.myetherwallet.com/js/vendors.307df816.js",
              "function": "Object.U.ptr.changeStatus",
              "in_app": true,
              "lineno": 161
            },
            {
              "colno": 3779858,
              "filename": "https://www.myetherwallet.com/js/vendors.307df816.js",
              "function": "Object.U.ptr.fireEvent",
              "in_app": true,
              "lineno": 161
            },
            {
              "colno": 82500,
              "filename": "https://www.myetherwallet.com/js/vendors.307df816.js",
              "function": "?",
              "in_app": true,
              "lineno": 161
            },
            {
              "colno": 33128,
              "filename": "https://www.myetherwallet.com/js/vendors.307df816.js",
              "function": "?",
              "in_app": true,
              "lineno": 57
            },
            {
              "colno": 35527,
              "filename": "https://www.myetherwallet.com/js/vendors.307df816.js",
              "function": "c.firmware",
              "in_app": true,
              "lineno": 57
            }
          ]
        },
        "mechanism": {
          "handled": false,
          "type": "onerror"
        }
      }
    ]
  },
  "platform": "javascript",
  "sdk": {
    "name": "sentry.javascript.browser",
    "packages": [
      {
        "name": "npm:@sentry/browser",
        "version": "5.15.5"
      }
    ],
    "version": "5.15.5",
    "integrations": [
      "InboundFilters",
      "FunctionToString",
      "TryCatch",
      "Breadcrumbs",
      "GlobalHandlers",
      "LinkedErrors",
      "UserAgent",
      "Vue"
    ]
  },
  "environment": "web",
  "release": "5.6.2",
  "event_id": "e8304f1673534d9b84a9a38f20b6887c",
  "request": {
    "url": "https://www.myetherwallet.com/access-my-wallet",
    "headers": {
      "User-Agent": "Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/83.0.4103.116 Safari/537.36"
    }
  },
  "tags": {
    "network": "",
    "service": "",
    "walletType": ""
  }
}
```

x1ddos, 2020-06-24:
Which MEW are you running?

x1ddos, 2020-06-24:
Also, which version of [bitbox-bridge](https://github.com/digitalbitbox/bitbox-bridge)?

x1ddos, 2020-06-24:
Actually, never mind https://github.com/digitalbitbox/bitbox02-firmware/issues/514#issuecomment-648977478 and https://github.com/digitalbitbox/bitbox02-firmware/issues/514#issuecomment-648987010.

From the report it seems the version is 5.6.2, and the bitbox-bridge is transparent proxy - irrelevant.

benma, 2020-06-24:
That is a bit of a surprise. I am testing it now and it seems to work. Let's figure it out:

Did MEW prompt you to upgrade, or the BitBoxApp?

Are you on myetherwallet.com or a local build?

Which firmware version are you on now? I am assuming v8.0.0.

x1ddos, 2020-06-24:
The 5.6.2 release of MEW seems to list and older version of bitbox package version in https://github.com/MyEtherWallet/MyEtherWallet/blob/8f4560de8cb9cf17b28416c053e3169afba5dbe9/package.json#L106:

```json
"bitbox02-api": "0.4.1"
```

Latest is 0.5.0.

x1ddos, 2020-06-24:
bitbox02-api v0.5.0 appears first in MEW release v5.6.3 in https://github.com/MyEtherWallet/MyEtherWallet/commit/e22b33c5468ca14dcd47373510f6b3c6bd9342b0.

x1ddos, 2020-06-24:
If it's coming from myetherwallet.com, why is not the latest v5.6.6-hotfix.2? According to https://github.com/MyEtherWallet/MyEtherWallet/releases/tag/v5.6.6-hotfix.2 anyway.

benma, 2020-06-24:
When I load myetherwallet.com I get the v5.6.6-hotfix.2 version. That one has [bitbox02-0.5.0](https://github.com/MyEtherWallet/MyEtherWallet/blob/v5.6.6-hotfix.2/package.json#L109), which would *not* prompt a firmware upgrade, but work anyway with BitBox02 v7.0.0 and v8.0.0.

@SCBuergel which MEW version does it show for you in the footer?

SCBuergel, 2020-06-26:
@benma - I just opened myetherwallet.com in my browser, don't know if I had a cached version but now it works.

Sorry for the pointless issue!

closed 2020-06-26 by SCBuergel

benma, 2020-06-26:
Nice!