BitBoxSwiss/bitbox02-firmware #526: Firmware 9.0.0 requested by Electrum Wallet

csh2000, 2020-07-04:
Hello, running Electrum Wallet AppImage 4.0.1 (just released). 

Connection to BB02 works fine, but it is asking for 9.0.0 firmware.

I have 8.0.0 installed and 9.0.0 seems not have been released yet...

Thanks!

[image]

benma, 2020-07-04:
We will release a new BitBoxApp with the firmware v9.0.0 in the coming days, restoring compatibility.

If it's urgent, please let me know, and we will figure out something quicker.

csh2000, 2020-07-05:
No problem, can wait.

Thanks!

closed 2020-07-05 by csh2000

benma, 2020-07-07:
The upgrade is out now, see the [blog post](https://medium.com/shiftcrypto/bitbox-07-2020-lidernen-update-493f670860aa) and the [app](https://shiftcrypto.ch/download/).

csh2000, 2020-07-08:
Hi, upgraded BitBox02  to 9.0 and worked fine with Electrum AppImage 4.0.2 in Linux Ubuntu 18.04.4 LTS.

Good point is that now I can use other "derivation paths", not only 84´/0´/0´!!

I think this could be included in BitBoxApp too.

Thanks!

benma, 2020-07-09:
You can vary only the last part, the account index, up to 100 accounts. The previous elements of the keypath are restricted, as a security measure.

> I think this could be included in BitBoxApp too.

Working on it :nerd_face: