BitBoxSwiss/bitbox02-firmware #578: Autocomplete of Bip39 words

My1, 2020-09-16:
The wordlist is designed in a way that all words are unique after 4 letters (in fact several backup solutions including the [Steelwallet](https://guides.shiftcrypto.ch/steelwallet/) only support entering 4 letters), so similar to on a keepkey you could just show the complete word instead for asking the extra letters.

while there are some words that are already unique on the third letter if that involves too much complexity one can ignore that.

deletion was brought up on telegram, one could for example check if the word currently has 4 or more letters to just cut to 3, and then proceed normally with deleting.

benma, 2020-09-23:
Voilà: https://github.com/digitalbitbox/bitbox02-firmware/pull/583/

This does autocompletion for now, with deletion working the the same as before, only one char at a time. Improving deletion is reserved for a future PR.

My1, 2020-09-23:
sadly no way for me to try :-(

closed 2020-09-24 by benma