BitBoxSwiss/bitbox02-firmware #608: Building on Windows

My1, 2020-10-03:
so currently the building process is really not made for windows so maybe it should someday be made a bit better.

I basically just downloaded the bitbox02-firmware-master folder to my D drive and installed docker and ran the docker commands manually and hacked them to make them work in a way with my limited knowledge.

~~~
docker build --pull --force-rm --no-cache -t shiftcrypto/firmware_v2 .

docker run --detach --interactive --tty --name="bb02-dev" -v "$repo_path":"D:\bitbox02-firmware-master\" --cap-add SYS_PTRACE shiftcrypto/firmware_v2 bash

docker exec -it "bb02-dev" bash
~~~

once I am inside the docker, the mounting didnt work for whatever reason so I needed to git clone yet again but I got a build with the proper hash.

there are a few problems tho.

1. the bitbox firmware directory didnt mount in
2. the bitbox02 cant mount into the docker for the bitbox cli tool

My1, 2020-10-17:
Update: I have a somehow working batch script for windows which is not pretty, but basically does all the important things:

https://gist.github.com/My1/66e0e0774f594225254150529019f066/

basically to get a quick run, place this into the scripts folder and run the following commands (file name doesnt really matter)

```
bb02-windows.bat image
bb02-windows.bat setup
```
to jump back into an existing running or stopped instance use

```
bb02-windows.bat start
```
using setup any existing docker containers called `bitbox02-firmware-dev` are overwritten. planning to amend this with a quickbuilder where one can literally specify a git URL and something to checkout (tag, commit, branch etc), so one can really quickly run a build for example on a PR branch (or reprod building), fully automated.

x1ddos, 2021-09-21:
hey @My1 are you still working on this? would be great to have building on windows working to widen dev platforms support.

My1, 2021-09-21:
I was working with this for a long time rather than on and just had a bunch of workarounds for pushing the firmware out to windows and stuff.