BitBoxSwiss/bitbox02-firmware #655: Hardware Design

StEvUgnIn, 2020-11-12:
Hi,
Where can I please find the source design of BitBox 02?

Is it open source hardware licensed under Creative Common International 4.0?

x1ddos, 2020-11-12:
Hey there. Have you already looked at the files in [doc](https://github.com/digitalbitbox/bitbox02-firmware/tree/master/doc) directory?

StEvUgnIn, 2020-11-13:
Sweet. But this directory does not contain design sources.

x1ddos, 2020-11-13:
Re: license. All files in this repository are published under the [LICENSE file in the root](https://github.com/digitalbitbox/bitbox02-firmware/blob/master/LICENSE), unless explicitely stated otherwise.

Which format would you like to see it in?
We have a plan to create a separate git repo just for the hardware design and publish it in KiCad. Would that be acceptable?

StEvUgnIn, 2020-11-14:
Yes, that would be greatly appreciated.

x1ddos, 2021-04-12:
I'll reopen to post here an update once we publish the hardware repo with all the sources.

mdhz, 2022-08-22:
Any update on this?
It'd be awesome if the gerber files were provided and a separate hardware repo was created.

benma, 2022-08-22:
cc @jadzeidan

mdhz, 2022-08-24:
> cc @jadzeidan

It seems, the dev is pretty inactive. 

No reply when tagged last time as well https://github.com/digitalbitbox/bitbox02-firmware/issues/655#issuecomment-948607471

Any others from the team can provide with the resources?

jadzeidan, 2022-08-24:
@mdhz apologize for the delay. I just need to check what we can publish exactly and will get back to you soon.

jadzeidan, 2022-09-14:
> > @mdhz apologize for the delay. I just need to check what we can publish exactly and will get back to you soon.
> 
> Ah ok. Will wait.

Hey @mdhz, sorry again to keep you waiting. The current Gerber files we have are for mass production and can't easily be used for one to create their own device (for example the OLED has a modified FFC so it's not an off-the-shelf part). 

We could make new Gerber files that makes it easier for DIY builders to make their own device (such as being able to use an off-the-shelf OLED) and it could also have a slightly different form factor, so malicious actors cant use it to produce exact copies of the PCB that is hard to distinguish from an original device. But this would take some time.

What do you you plan to do specifically? If you want to load your own firmware onto the device, we would be happy to send you a developers version of the device. Just email support@shiftcrypto.ch and we can help you from there.

mdhz, 2022-09-19:
> > > @mdhz apologize for the delay. I just need to check what we can publish exactly and will get back to you soon.
> > 
> > 
> > Ah ok. Will wait.
> 
> Hey @mdhz, sorry again to keep you waiting. The current Gerber files we have are for mass production and can't easily be used for one to create their own device (for example the OLED has a modified FFC so it's not an off-the-shelf part).
> 
> We could make new Gerber files that makes it easier for DIY builders to make their own device (such as being able to use an off-the-shelf OLED) and it could also have a slightly different form factor, so malicious actors cant use it to produce exact copies of the PCB that is hard to distinguish from an original device. But this would take some time.
> 
> What do you you plan to do specifically? If you want to load your own firmware onto the device, we would be happy to send you a developers version of the device. Just email [support@shiftcrypto.ch](mailto:support@shiftcrypto.ch) and we can help you from there.

Hey, no worries.

I was planning to make my own device since I've tried it with Trezor One already as it was pretty easy as gerber files were readily available. 
I'd appreciate it if you can provide the gerber files for DIY enthusiasts (also the 3D outer casing similar to Trezor) if it has modified FF and also the BOM and part number.

[image]

jadzeidan, 2022-09-19:
> Hey, no worries.
>
> I was planning to make my own device since I've tried it with Trezor One already as it was pretty easy as gerber files were readily available. I'd appreciate it if you can provide [...]

Yes, this is something I would like to do as well. Publish a DIY version of the gerber files along with a 3D printable case so people can make their own device. Resources are tight now so it probably wont be available soon but I'll keep this issue updated about any progress :)

mdhz, 2022-09-19:
> Yes, this is something I would like to do as well, publish a DIY version of the gerber files along with a 3D printable case so people can make their own device. Resources are tight now so it [...]

Oh OK :)

StEvUgnIn, 2023-01-31:
Should we close this issue?

mdhz, 2023-01-31:
> Should we close this issue?

Still no DIY gerber files?

If no, then yeah close it.

jadzeidan, 2023-01-31:
> > Should we close this issue?
> 
> Still no DIY gerber files?
> 
> If no, then yeah close it.

No update since the last comment. So yes we can close the issue and re-open if there is any progress.

benma, 2023-02-01:
> So yes we can close the issue and re-open if there is any progress.

This is not how issues work :joy_cat: Just leave it open until done so others can find it, or close it if there is no more interest.

keon, 2023-02-27:
Also waiting for this to be released!

dylangerdaly, 2023-12-21:
For anyone wanting to run their own firmware on a Bitbox02, check out https://github.com/digitalbitbox/bitbox02-firmware/issues/1157

I'll try use my ChipSHOUTER to get the MCU back into a debuggable state, that should let me re-flash firmware with my own keys, I'll update via the email and issue linked above.

Aside from that I'd very much like to build my own device, another idea I had was to manually re-work the MCU (ATSAMD51J20A) from the board, that would mean de-soldering the MCU and replacing it with an untouched new MCU.

@jadzeidan is the PCB epoxied?

Sami32, 2025-07-02:
In the mean time while waiting it for the coming years, until they release an open source hardare with PCB schematic and Autocad files, some alternative reading dreaming that one day BitBox will also be added on their webpage:

https://github.com/diybitcoinhardware/diybitcoinhardware.github.io