BitBoxSwiss/bitbox02-firmware #657: Encrypted microSD card backup

nioncode, 2020-11-15:
It would be great if the backup to the microSD card could be encrypted with a user defined password to prevent giving attackers full access to funds in case they get hold of the SD card (and no BIP39 passphrase is used).

It probably makes more sense to type the password on the PC doing the backup to allow the usage of a password manager to enforce a strong password instead of entering it on the BitBox02 itself. If we assume that the PC is compromised, the attacker could steal the password, but if they did not get a copy of the SD card before (which seems quite unlikely), they can not do anything with the stolen password, since the SD card is inside the BitBox and not on the pc itself, so I'd say the risks are low compared to the gained UX (entering a 20 char password on the BitBox is a bit tedious). However, if you decide otherwise, entering on the BitBox itself would also be ok I guess, but will most likely result in weaker passwords.

Proposed workflow:
- after pressing 'Create Backup' in the app, there should be a checkbox 'Encrypt Backup'
- when checked, a text field allows typing in the password (optionally a second text field to confirm the password? Although I'd argue that this is unnecessary since user's verify their backup afterwards)
- when confirming the backup creation, the password gets sent to the BitBox, which encrypts the backup content (which cipher to use? I haven't check which ciphers are available to use on the BitBox)

When verifying or restoring a backup, the BitBox should detect that a password is required and prompt for it in the app / on the screen, then decrypt the backup and verify / restore as before.

benma, 2020-11-15:
Thanks for the request. This feature is requested quite often, though we have been hesitant to add it due to the great potential of loss of funds, as user mistakes with passwords are far more common than physical theft). We definitely keep track of this feature and revisit it every once in a while.

In the meantime, you can use the optional passphrase feature to better protect your backups if you really need to. Note however that it is an expert feature and requires great care: 

https://guides.shiftcrypto.ch/bitbox02/advanced/passphrase/#risks-of-using-a-passphrase

nioncode, 2020-11-15:
The problem with the BIP39 passphrase is that you need to enter it every time you use the device, which is bad from a UX perspective.

If you don't want to show the encrypted backup as prominently in the ui, maybe the checkbox should only be shown if the user activated encrypted backups in the expert settings (just like the BIP39 passphrase feature).

My1, 2020-11-30:
that would certainly be something better than nothing, this thing would also need a really serious warning at least in my opinion similar to the passphrase one but a bit more strongly

nioncode, 2020-11-30:
Shamir backup is tracked in #113 (for paper based backups though). I don't think it makes sense to have a 2-of-3 shamir backup on microSD since you'd need 3 microSD cards, but from a UX perspective this would be great, since to recover your funds you just have to plug in two SD cards after another and don't need to know any passphrase / seed, which is great for regular users.

My1, 2020-11-30:
totally it would be very easy. depending on how much the BB02 can store it could even support restoring them one by one seperately so that the backups dont have to "come together" in a vulnerable state.

alex27riva, 2021-04-25:
Hi everyone, is possible to move the backup file on a Veracypt container / partition? (using a live environment like Tails OS)
Or the backup file is linked in some way to the micro SD card?

benma, 2021-04-26:
@alex27riva it is in theory possible to copy the files out of the microSD card into a different secure environment. However, it might be better to avoid generic purpose computers to handle your secrets. Alternatively, you could make redundant backups on multiple microSD cards, or also to write down the backup in the form of [24 recovery words](https://shiftcrypto.support/help/en-us/20-24-recovery-words). This will also make the recovery process a lot easier and safer.

wbkg, 2022-01-24:
I am considering making multiple copies of the micro SD card (using duplicate cards I purchased directly from Shift Crypto). My question is if there is any way to copy the backup file from one card to another in an offline environment. I am reluctant to do so on a computer even if wifi is turned off because it might be possible for the copying operation to be spied on, logged or captured, then exposed when the computer reconnects to wifi and goes online.

My1, 2022-01-24:
> My question is if there is any way to copy the backup file from one card to another in an offline environment

theoretically you can but it is that secure as you mean.

> The perfect feature would be one that allows the BitBox02 to save the backup file on more than 1 card. After setting up the wallet, this is unfortunately not possible.

that's actually wrong.
manage device -> manage backups -> create backup.

thisconnect, 2022-01-24:
@wbkg you can use the BitBox02 device to write the backup to multiple microSD cards and also check if the backups are valid.

wbkg, 2022-01-24:
I see. Thank you very much for this helpful tip. Will do it on my BitBox02 when I'm back home from work. I'm totally paranoid about security after a friend lost 0.85 BTC last week. I ordered a second BitBox02 and this Crypto Capsule as backups and started using a passphrase on top of the 24 words.

benma, 2022-01-24:
@wbkg can you share some info about how the .85 BTC were lost (without revealing any sensitive information)?

> and started using a passphrase on top of the 24 words.

Be sure to read and understand the passphrase and its risks: https://shiftcrypto.support/help/en-us/21-optional-passphrase

wbkg, 2022-01-24:
@benma the 0.85 BTC were on his Trezor 1 and lost by negligence. My friend didn't use a passphrase and he had a habit of storing everything on his Google Keep app and also emailing it to himself. His Google has Authy 2FA (Twilio app) on a Surface Pro device. He also lost a small amount of ETH. 

What I'm hoping to see is the BitBox app being accessible in other android devices too. I use a Chromebook with Android 11 integration. When I try to download the app from the Google Playstore I get the message "BitBoxApp by Shift Crypto isn't available on Google Play on this device" which is unfortunate. I have Linux on this Chromebook and I did install the BitBox app on it but it wasn't detecting the device. I am not very good with Linux commands so I ended up uninstalling it, fearing I might do something wrong along the way.

My1, 2022-01-25:
chromebooks are iirc not good yet with USB stuff, so that might be annoying. one thing you could be able to try would be if chromebooks at least do webusb/webHID by trying My Ether Wallet or adalite (provided your firmware is recent enough) to see if you can connect the bitbox to it