BitBoxSwiss/bitbox02-firmware #674: "Error fetching Backups" when changing microSD card in the "Manage backups" screen

cstenglein, 2020-11-29:
To reproduce:

1. Insert microSD
2. Go to "Manage devices" -> "Manage backups"
3. remove microSD and insert another microSD
4. Since the screen doesn't update, click "Back"
5. Go to "Manage backups" again

[image]

The error also occurs when going back to the "Manage devices" screen and changing the microSD.

Workaround: Remove microSD & Restart of the Bitbox02 App. Restating with microSD inserted will continue to show the error.

benma, 2020-11-29:
Thanks for the report.

> remove microSD and insert another microSD

Does it have to be another microSD card? What happens if it is the same?

cstenglein, 2020-12-01:
Hi,

if it's the same card it depends:

- Insert SD -> Manage Backups -> Press "OK" -> Remove SD -> Back -> Insert SD -> Manage Backups -> Press "OK" Works at first but if I remove the SD and click on "Manage Backups", I get the error again. If I insert the SD and then click on Manage backups, the "OK" Button is not clickable, only back.

- Insert SD -> Manage Backups -> Press "OK" -> Back -> Remove & reinsert SD -> Manage backups leads to the error described in my post from 2 days ago.
 

The error log only gave me this (other things not deemed relevant; if you need more of the log, just write it here): 

>time="2020-12-01T17:19:03+01:00" level=debug msg="List backups " device=bitbox02 group=handlers

> time="2020-12-01T17:19:03+01:00" level=warning msg="Received an error from Bitbox02" bitbox02-error=103 device=bitbox02 group=handlers

benma, 2021-02-23:
Sorry for the wait. This seems to be a BitBoxApp issue, not a firmware issue. It should be fixed with this PR and land in one of the next releases:

https://github.com/digitalbitbox/bitbox-wallet-app/pull/1188

closed 2021-02-23 by benma

mclauncherlinux, 2024-01-07:
[image]