BitBoxSwiss/bitbox02-firmware #701: Housing specs

unixb0y, 2021-01-07:
Would it be possible to get exact housing dimensions to design 3D-printable cases for the BitBox02?

x1ddos, 2021-01-10:
@jadzeidan should be able to produce more precise metrics compared to https://shiftcrypto.ch/bitbox02/#specifications.

jadzeidan, 2021-01-24:
@unixb0y Here are dimensions of the PCB and the location of the OLED with respect to it:

[image]
[image]
[image]

unixb0y, 2021-01-24:
@jadzeidan Thanks a lot!

closed 2021-01-24 by unixb0y