BitBoxSwiss/bitbox02-firmware #727: Signing a message with a Bitcoin address?

lf94, 2021-02-25:
Is there any support for signing a message with a Bitcoin address? If not, what's the rationale?

benma, 2021-02-25:
It is supported. You can sign messages using Electrum (right click an address, sign).

See also: https://shiftcrypto.ch/blog/bitbox02-with-electrum-wallet/

closed 2021-02-25 by benma

lf94, 2021-02-25:
Thank you for the reply @benma :slightly_smiling_face: 

I asked because it appears bitbox02-api-js doesn't expose this functionality (https://github.com/digitalbitbox/bitbox02-api-js/issues/35). At the very least, at least it appears I can add support since the underlying firmware supports it.

x1ddos, 2021-02-26:
@lf94 JS API does support it. Let's continue in that issue.