BitBoxSwiss/bitbox02-firmware #728: add support for Cardano

airtonix, 2021-03-04:
nuff said really. Ada is awesome.

benma, 2021-03-04:
Thanks for the request. We currently don't have any active plans for adding more coins at the moment. If that should change, we'll evaluate ADA.

closed 2021-03-04 by benma

airtonix, 2021-03-11:
@benma Um... if this is truly opensource, you'd leave the ticket open and tag it as "good first feature" ? 

Like I'm looking at the `src/apps` directory and, I dunno I"m not a c/c++ developer... but it kinda looks like the system is setup to have more coins supported.

Maybe someone else could have written that new feature, but by closing this ticket you've effectively communicated to the rest of us that "No, contributions are not welcome"

airtonix, 2021-03-11:
Gotta be honest here, pretty disappointed in this product now.

benma, 2021-03-11:
I am sorry to hear that. Generally, we are of course happy about contributions. Open source means that you can inspect the code, copy it, modify it, build on it, and so on - it does not mean we can or will accept any feature or contribution. Especially big additions require coordination and the blessing with the maintainers, as with any open source project. 

Adding a coin is not a one-shot contribution either - it requires infrastructure, BitBoxApp and/or third party wallet integration, customer support, maintenance, and so on. 

That being said, we might add more coins in the future, and ADA will certainly be in the list of candidates for inclusion.