BitBoxSwiss/bitbox02-firmware #732: Denying incorrect date on wallet recovery setup results in complete failure

airtonix, 2021-03-11:
following this: https://shiftcrypto.support/help/en-us/20-24-recovery-words/48-how-to-restore-an-externally-generated-wallet-on-a-bitbox02

1. Step 7, the device shows an incorrect date/time (out by half an hour).
2. select no.
3. device shuts down. 😶 🙄

soooo... what? I restart the whole dance again, only to be shown an incorrect date again? i mean.... what's the point? there is a point right?

benma, 2021-03-11:
Hi

What do you mean by 'device shuts down'? Declining the date should result in the BitBox02 going back to the default screen and being responsive to actions in the app.

As to your issue: the date is off roughly by the amount of time you spent entering the recovery words. For now, if the time is off by half an hour or so, you can simply proceed (accept) at no risk.

That being said, we already fixed this issue, and with the next release, the date/time confirmation is going to happen before entering the recovery words, which should make it accurate again.

Apologies for the inconvenience. Let me know if there are any further issues.

benma, 2021-03-11:
For reference, here is the commit fixing this issue: https://github.com/digitalbitbox/bitbox02-firmware/pull/690/commits/9ed4323ed8eab2c4aa8e2ce272201b1962bca165

airtonix, 2021-03-16:
> As to your issue: the date is off roughly by the amount of time you spent entering the recovery words. For now, if the time is off by half an hour or so, you can simply proceed (accept) at no risk.

It was off by half an hour, we only spent a few minutes entering the words.

I feel like we'll be revisiting this issue later.

airtonix, 2021-03-16:
> What do you mean by 'device shuts down'? Declining the date should result in the BitBox02 going back to the default screen and being responsive to actions in the app.

I mean it shuts down. we have to unplug and replug to start the whole dance again.

benma, 2021-03-16:
> It was off by half an hour, we only spent a few minutes entering the words.

The time is sent by the host computer / mobile wallet. Is the time there correct?

If you want, you can already flash the next version v9.5.0 using py/load_firmware.py: https://github.com/digitalbitbox/bitbox02-firmware/releases/

Or wait until next week when we will release it along with the BitBoxApp.

Then we can see if this helps, and if not, at least we can skip entering the 24 words when debugging.

x1ddos, 2021-04-12:
> Or wait until next week when we will release it along with the BitBoxApp.

This was released in [firmware v9.5.0](https://github.com/digitalbitbox/bitbox02-firmware/releases/tag/firmware%2Fv9.5.0), bundled in [app v4.27.0](https://github.com/digitalbitbox/bitbox-wallet-app/releases/tag/v4.27.0).

> I mean it shuts down. we have to unplug and replug to start the whole dance again.

@airtonix I can't reproduce it. Tried with the latest firmware/app version linked above. The device goes to the default screen like @benma said.

@airtonix can you post a short video or some screenshots/photos of what's happening in your case?

Xzyer, 2021-05-14:
I'm having a similar issue with my new BitBox02 Multi, except my device doesn't power down. I am using the mobile app and the time is 1 day in the future. There doesn't appear to be any way to change it in the app and my system time in android is correct.

benma, 2021-05-14:
@Xzyer what timezone are you in? Was the time correct, but just the date day was off by one, or was the time also incorrect? Did you start the recovery workflow near midnight? 

Even though the date/time is a bit off, you can proceed with no risk: the date/time is used to calibrate U2F, and a date in the near future is okay.

Xzyer, 2021-05-15:
> @Xzyer what timezone are you in? Was the time correct, but just the date day was off by one, or was the time also incorrect? Did you start the recovery workflow near midnight?
>
> Even though the [...]

I'm in UTC -6. I was doing the setup around 6pm and it said it was the next day, it didn't display a time, just a date.

benma, 2022-10-04:
Fix for the time being shown wrong on mobile: https://github.com/digitalbitbox/bitbox-wallet-app/pull/1789

closed 2022-10-10 by benma