BitBoxSwiss/bitbox02-firmware #743: Feature: Dasher style "keyboard"

Giszmo, 2021-04-11:
The text input is understandably very cumbersome but I think a [dasher](https://www.youtube.com/watch?v=2Z7fr4pZhmQ) style input could work very intuitively and less disorienting.

Currently, to type a "t" I have to type right-left-right but it's not initially clear that the second click goes left and the third right after having found "t" on the initial screen. The user can't follow those quick slides of the letters, at least not in the first transition from full ABC. With the proposed change, he would

* click-and-hold somewhere near the "t"
* the alphabet would smoothly spread out from where his touch is being detected
* if he touched left of the "t", "t" would travel right
* confused users would release the touch but still wanting "t", they would return to touch around "t", resuming the "zoom" effect
* as soon as "t" is the only letter on the screen, zooming would stop
* releasing would chose "t"

The layout should get a slight change that the current input method already would benefit from:

Instead of

```
a b c d
e f g h
```

show

```
a c e g
 b d f h
```

The current easing from the two-line layout to the "a b c    d e    f g h" comes with letters running other letters over wildly which makes it harder to follow for the eye. "e" is for example covering the "|" of the "d" for some frames.

The new easing could delay the first line sinking into the second line, avoiding all overlap. After all, the full alphabet almost fits in one line already.

jadzeidan, 2021-04-18:
Thanks for the detailed proposal. Password input it definitely the trickiest part of using the BitBox02, whish is expected because of how "involved" the process compared to other on-device interactions. We tested quite a few password input methods before and found that our "ternary" method was the easiest to use while being the most precise as well (less wrong inputs than other methods). Of course this can always be improved. I haven't seen the dasher style input before. I tried out this [demo](https://dasher-project.github.io/redash/browser/), but I admit it was quite tricky to use :p. I think the only way to know for sure if it is an improvement is to test it.  Maybe this is something we could test in the future when we have more time and resources.

Giszmo, 2021-04-19:
The demo you found lacks some configurability to really appreciate what I had in mind.

* it scrolls at dynamic speed which would not be possible on BB2
* it scrolls way too slowly. As the full ABC would be possible when starting to write, it could zoom at least twice of the demo's top speed.
* has all the letters, capital and small and special characters in the list which should remain in the switch above as is now
* it has predictions switched on which naturally would not make sense on BB2. This you can switch off though

Giszmo, 2021-04-20:
I put out a [little bounty to implement this idea as a demo in JS](https://twitter.com/LeoWandersleb/status/1384327997361213445). I'm curious to try it out. Actually I'm tempted to write it myself but I'm not a JS guy and I suspect it's more complicated than it looks at first.

Of course in the BB2 firmware it's orders of magnitude more complicated to implement which makes such a JS demo all so more valuable I hope.