BitBoxSwiss/bitbox02-firmware #744: Improve password input security

Giszmo, 2021-04-11:
If entering the password in a public setting, it is almost impossible not to leak it. The discreet left-middle-right clicks are very easily detected at a great distance.

Although it makes input even harder, the three groups should get shown in random order.

(Should my proposal #743 get implemented, then the alphabet could start looping at a random pace until click occurs. In this mode, the initial click may be very imprecise.)

jadzeidan, 2021-04-18:
Hey @Giszmo, thanks for the input (pun intended :p). I understand the concern, however something like this will likely have a strong negative impact on usability, which a key component of the BitBox02 design principle (to be the easiest to use hardware wallet). Also, the screen would still be visible to the public, so I think the overall benefit of a randomizing the characters is limited. If something like this were to be implemented, it would probably be an advanced option.

malesch, 2021-06-29:
I was always wondering if the slider input of the BB would allow scrolling directly through the list of input characters (or block of chars: a-z,A-Z,0-1,...) what would allow much faster input and improve security. The current method requires multiple interactions just to select a single char. In my opinion this is not very user friendly and tempts to limit yourself to shorter phrases.

benma, 2021-06-29:
@malesch it would be possible, and ideally we'd like to offer both ways of input.

We tested some prototypes of inputting via scrolling in the past, and while it was a bit more intuitive, it also turned out to be a slower way of entering with the prototypes we made. The reason is that tapping one of the three groups can be done quickly and accurately and committed to muscle memory. 

That being said, we didn't iterate a ton on the scrolling variant, so there is room for improving that too, maybe even to the point where it can be as quick as the tapping variant. However, it was always low priority compared to other features since the current way of inputting seems to work very well generally. It is not likely we will work on the scrolling variant anytime soon.

cc @jadzeidan