BitBoxSwiss/bitbox02-firmware #746: Error: panicked at after unlocking

voyager-dotnet, 2021-04-30:
Hey,

I setup my new bitbox. After setting it up, I unplugged and replugged it.
I entered the password to unlock the device and got the following error:

```
Error: panicked at `keystore unlock failed`, src/bitbox02-rust/src/workflow/unlock.rs:87:14`
```

[image]

After I unplugged and tried again, it worked. Could not reproduce it a 2nd time.
Do I have to be worried? Any idea why this happened?

thisconnect, 2021-04-30:
I believe this happened to me too on an Android with wonky usb-c connection when the device power got interrupted.

benma, 2021-05-01:
Thanks for the report @voyager-dotnet . Does it happen everytime you unlock, or was it only one time?

benma, 2021-06-08:
Adding more information to this error message so can have more to go on if it happens again: https://github.com/digitalbitbox/bitbox02-firmware/pull/757

benma, 2023-10-27:
We have improved the communication to the securechip by switching to polling instead of using timings. Closing as there is nothing more to do for now.

closed 2023-10-27 by benma