BitBoxSwiss/bitbox02-firmware #754: BitBox Gnosis Safe Cosigner Integration With Blocknative Onboard.js

AdamSHurwitz, 2021-05-31:
Hi BitBox Team,

Are there plans to work with Blocknative in order for them to add support for the BitBox SDK in the [Onboard.js library][1]?

Once completed, the [Gnosis team stated][2] the BitBox will be a great option as a cosigner in a Gnosis Safe wallet because it will be able to [verify the `safeTxHash`][3] along with the Trezor, Ledger, and GridPlus hardware wallets.

[image]

Much appreciated!

[1]: https://github.com/blocknative/onboard
[2]: https://discord.com/channels/477106835862716416/477391201708802058/847876036405493760
[3]: https://help.gnosis-safe.io/en/articles/4818770-how-to-verify-gnosis-safe-transactions-on-a-hardware-wallet

benma, 2021-05-31:
Would it be possible for the Gnosis team to use our open source JS library to add support in Onboard.js? https://www.npmjs.com/package/bitbox02-api

cc @jadzeidan

AdamSHurwitz, 2021-05-31:
From what Gnosis has informed me, Blocknative must add a BitBox SDK to the Onboard.js library in order for the hardware wallet to act as a cosigner.

@jadzeidan, followed up via Telegram. Thank you!

[image]

benma, 2021-05-31:
What does that mean, add a BitBox SDK? From the sound of it. Onboard.js is simply a library that talks to multiple hardware wallets? So someone (possibly Gnosis/BlockNative?) can use our Javascript library and integrate.

AdamSHurwitz, 2021-05-31:
@benma, Onboard.js is a library that provides wallet integrations to developers. Gnosis Safe uses this library to confirm Ethereum transactions.

That would be fantastic if Blocknative can integrate the BitBox Javascript library. From my understanding, the current [wallets that are a part of the library][1], have worked with Blocknative to integrate this.

[1]: https://docs.blocknative.com/onboard#wallet-modules

benma, 2023-10-27:
Closing as there is nothing to do here in the firmware repo. Gnosis/Onboard can integrate using our libraries, see https://bitbox.swiss/dev/.

closed 2023-10-27 by benma