BitBoxSwiss/bitbox02-firmware #762: What is the pre-image of the hash in the release notes?

Giszmo, 2021-07-07:
https://github.com/digitalbitbox/bitbox02-firmware/releases lists binaries and hashes but I couldn't find a match.

So the device checks firmware hashes prior to installation? It should ask something like "Do you want to install the validly signed firmware update version X with hash Y?" so I can then check if that's really what I want, right? So regardless where the companion app pulled the update, if I pulled it from above link and verified both hash and reproducibility, I could approve the known hash but I only see a "random" hash. Where is the missing link?

benma, 2021-07-07:
Hi

The device only boots firmware that was properly signed, so flashing incorrect firmware does nothing, and one can always restore valid firmware afterwards too.

As for the hash shown in the release notes, please check this documentation: 

https://github.com/digitalbitbox/bitbox02-firmware/tree/master/releases#verify-the-hash-as-shown-by-the-bitbox02-at-startup

In summary, the hash you are seeing is the `sha256d(<version><padded firmware>)`, and there is a small, easy to audit tool in the same folder to help reproduce it:

https://github.com/digitalbitbox/bitbox02-firmware/blob/master/releases/describe_signed_firmware.py

The version is part of the hash because of downgrade prevention.

The bootloader of the device can be configured to display the hash of the version+firmware that is currently installed *before* booting the firmware as an additional help.

I hope this helps. Please let me know if we can help improve the process or documentation. I already added the above link to the release notes to make it easier to discover.

Giszmo, 2021-07-10:
So I figured out how to extract the version but how would I know the version without extracting it from the signed firmware file?

Does the BitBox02 also show the hash of the whole file? This would give assurance that the companion app downloaded the correct file others reproduced and approved as reproducible. Also does it show the version and which signatures are found? All that would be kind of important with the verifiable hash not covering version and sigs.

I had the above questions for my [review of the BitBox02 for WalletScrutiny](https://walletscrutiny.com/hardware/bitBox2/) which now is the first hardware wallet we reviewed and also attested reproducibility.

benma, 2021-07-10:
The version in this context is a monotonically increasing number, and it is part of the hash that is signed to prevent downgrades (you cannot run a firmware with a lower monotonic version than what you had previously on the device). For example, the current number for the BTC-only firmware is 22.

You don't need to know the actual value though, at least I don't see how it is helpful. If you want to know it anyway, you can read it from the signed firmware releases using the describe_firmware.py tool, and convince yourself that they increase with each release.

> Does the BitBox02 also show the hash of the whole file?

If you enable showing the hash before boot, it shows the hash of `<version><padded firmware file>`, but that is enough information to link it to your reproducible build using describe_firmware.py as explained above.

>  Also does it show the version and which signatures are found? All that would be kind of important with the verifiable hash not covering version and sigs.

The hash contains the version and that hash is signed, producing the signatures. so the hash can't contain the sigs. Showing the actual signatures does not add any benefit - the firmware only boots if the signatures verifies against the hash.

To elaborate, the firmware+version must be properly signed, otherwise the firmware won't boot, or the bootloader itself is fake, in which case you can't trust what it tells you anyway. 

Still, the BitBox02 bootloader has the ability to return and display the hash of the signing pubkeys (can be done e.g. via this [tool](https://github.com/digitalbitbox/bitbox02-firmware/blob/780522b1d0b32c17046ff5c66df882d725bdb31f/py/send_message.py), and you can extract the actual signatures from the signed firmware file and match them to the pubkeys if you really desire - though that would be quite a bit of work and we haven't documented this workflow properly, as the bootloader essentially performs this work.

benma, 2021-09-07:
Since the [review concluded](https://walletscrutiny.com/hardware/bitBox2/), I will close this issue as it seems to have resolved  the questions. Please re-open if needed.

Giszmo, 2021-10-04:
The issue is not resolved. I still fail to understand why the hardware wallet should not display the full signed file's hash. The bootloader checking a signature which commits against yet another hash is not really relevant here if I primarily want to make sure the file the bootloader sees is the one I audited.

I now would have settled with running the test script you provide, following the logic that it's part of the code that needs to be reviewed and determined to be benign, which so far is not my focus but I fail to run the script.

The release notes say: "See [this documentation](https://github.com/digitalbitbox/bitbox02-firmware/tree/firmware-btc-only/v9.7.0/releases#verify-the-hash-as-shown-by-the-bitbox02-at-startup for more details about this hash." and there it says "run `./describe_signed_firmware.py`" which errors out saying "read py/README.md" which says "run `pip3 install` in the bitbox02 directory" but I get "ERROR: You must give at least one requirement to install" running this in the three more likely correct of the 7 "bitbox02" directories.

benma, 2021-10-04:
I can look into replacing this script with a minimal Python script without dependencies that does the same thing. 

In the meantime: `pip3 install .` inside of `py/bitbox02` should work. Note the `.` dot after `pip3 install`. The error you mention happens if you missed the dot.

benma, 2021-10-04:
@Giszmo here is a draft PR. Please try running the script inside.

benma, 2021-12-02:
Closing this again since the above PR is now merged and the describe_firmware.py tool has no dependencies outside of the standard library. Please re-open if needed.

closed 2021-12-02 by benma

x1ddos, 2021-12-02:
@Giszmo so, to retiterate what @benma described above while looking at the python script: any signed firmware.bin file posted in a release assets is

    magic marker bytes + signature + version + firmware

the bootloader displays a hash of `(version + padded firmware)`. if you rebuild the firmware from sources, pad it with `0xff`, take sha256 twice, it should always match what the bootloader shows on the screen, while `sha256(unpadded firmware)` should match what's in assertion.txt files.

Giszmo, 2021-12-02:
@x1ddos as I said above, I had that part figured out and will not rely on Python tools going forward as my bash script does the job fine already. More details [here](https://walletscrutiny.com/hardware/bitBox2/) or to save you the click:

----

We wrapped the findings from prior reviews in a [test script](https://gitlab.com/walletscrutiny/walletScrutinyCom/-/blob/master/scripts/test/hardware/bitBox2.sh) which gave us these results:

```
$ scripts/test/hardware/bitBox2.sh 9.8.0
...
Hashes of
signed download             51e584b61eaff83fe9e538c0fd47c617c686cd6da1a501acc33f12dab37f627a  firmware-btc.v9.8.0.signed.bin
signed download minus sig.  a397bb27eda4da1b3fb8e7462b26b4fc49afa15c00cf75e3e45983ddd533a7e2  p_firmware.bin
built binary                a397bb27eda4da1b3fb8e7462b26b4fc49afa15c00cf75e3e45983ddd533a7e2  temp/build/bin/firmware-btc.bin
hash as shown in bootloader abd2c1f47d6e5f49e67a51f163de87b83b080a8d65d12a62f21b9d5517a66cd1
                            (The latter is a double sha256 over version, firmware and padding)
```

So, the result looks good. The tested firmware is **reproducible**.

x1ddos, 2021-12-02:
@Giszmo very nice. would you care to contribute a signature following https://github.com/digitalbitbox/bitbox02-firmware/tree/master/releases#contribute-your-signature :pray:

Giszmo, 2021-12-03:
It is on my todo list to provide pgp signed statements for the verdicts shown, together with a system where multiple experts may provider conflicting verdicts but I'm not sure yet how to present that on the website or how to get people to put in the work to re-build binaries. For now I have another 6 hardware wallets waiting to be rebuilt :( / :)