BitBoxSwiss/bitbox02-firmware #799: Too many compile warnings when make unit-test

x1ddos, 2021-09-21:
These make it harder to spot important messages:

```
src/ui/ugui/ugui.c: In function 'UG_MeasureStringCentered':
src/ui/ugui/ugui.c:622:45: warning: field precision specifier '.*' expects argument of type 'int', but argument 4 has
 type 'long int' [-Wformat=]
             snprintf(line, sizeof(line), "%.*s", c - start, start);
                                           ~~^~   ~~~~~~~~~
src/ui/ugui/ugui.c:630:37: warning: field precision specifier '.*' expects argument of type 'int', but argument 4 has
 type 'long int' [-Wformat=]
     snprintf(line, sizeof(line), "%.*s", c - start, start);
                                   ~~^~   ~~~~~~~~~
src/ui/ugui/ugui.c: In function 'UG_PutStringCentered':
src/ui/ugui/ugui.c:763:75: warning: field precision specifier '.*' expects argument of type 'int', but argument 4 has
 type 'long int' [-Wformat=]
             snprintf(lines[current_line], sizeof(lines[current_line]), "%.*s", c - start, start);
                                                                         ~~^~   ~~~~~~~~~
src/ui/ugui/ugui.c:768:67: warning: field precision specifier '.*' expects argument of type 'int', but argument 4 has
 type 'long int' [-Wformat=]
     snprintf(lines[current_line], sizeof(lines[current_line]), "%.*s", c - start, start);
                                                                 ~~^~   ~~~~~~~~~

src/ui/components/confirm.c: In function 'confirm_create':
src/ui/components/confirm.c:108:58: warning: format '%u' expects argument of type 'unsigned int', but argument 4 has
type 'size_t' {aka 'const long unsigned int'} [-Wformat=]
         snprintf(size_label, sizeof(size_label), "Size: %uB", params->display_size);
                                                         ~^    ~~~~~~~~~~~~~~~~~~~~
                                                         %lu

src/apps/eth/eth_sighash.c: In function 'app_eth_sighash':
src/apps/eth/eth_sighash.c:117:5: warning: nested extern declaration of 'rhash_keccak_final' [-Wnested-externs]
     rhash_keccak_final(&ctx, sighash_out);
     ^~~~~~~~~~~~~~~~~~

test/unit-test/test_memory_functional.c: In function '_test_memory_multisig_full':
test/unit-test/test_memory_functional.c:113:53: warning: format '%ld' expects argument of type 'long int', but argume
nt 4 has type 'size_t' {aka 'long unsigned int'} [-Wformat=]
         snprintf(names[i], sizeof(names[i]), "name%ld", i);
                                                   ~~^   ~
                                                   %ld
```

benma, 2021-09-27:
> src/apps/eth/eth_sighash.c: In function 'app_eth_sighash':
> src/apps/eth/eth_sighash.c:117:5: warning: nested extern declaration of 'rhash_keccak_final' [-Wnested-externs]
>     rhash_keccak_final(&ctx, sighash_out);
>     ^~~~~~~~~~~~~~~~~~

fixed by https://github.com/digitalbitbox/bitbox02-firmware/pull/796

benma, 2021-09-27:
Updated the PR to fix/hide these as well

closed 2021-09-27 by benma