BitBoxSwiss/bitbox02-firmware #807: xch / Chia support

chriseomi, 2021-10-03:
Would be nice to see.

https://old.reddit.com/r/chia/comments/pfpjlx/hardware_wallet_support/

Currently not supported by any hardware wallet as it needs BLS support.

https://en.wikipedia.org/wiki/BLS_digital_signature

So might be hard to bring to BitBox02

jadzeidan, 2021-10-04:
@chriseomi thanks for the request. Adding more altcoins takes up a lot of resources and requires long term maintenance. XCH seems to be very new so we will need to wait and see how it develops before we can consider adding it.

WatcharinNate, 2021-12-03:
0000000000000000000000000000000000000000000000000000000000000000000010000000612552827027247782069571266955955800000100000000000000000005032534822353401010000100010000002321860128077044238488945017400000000380100000002393092232634000000000930000000000010041166163

benma, 2023-05-29:
Duplicate: https://github.com/digitalbitbox/bitbox-wallet-app/issues/1444

We are not planning to add any more coins for the time being.

closed 2023-05-29 by benma