BitBoxSwiss/bitbox02-firmware #825: Feature: show BitBox name on lock screen

felix-iw, 2021-11-03:
As a user of multiple BitBox hardware wallets, I would like to see the name of the BitBox on the lock-screen so that I can easily identify the device before I enter the password.

Reason: every BitBox looks the same. Stickers are currently the only way to put a label that is visible before unlocking.
But stickers may peel off.
As a user of multiple Bitboxes, I run the risk of mixing up the devices and entering incorrect passwords.
An optional display of the device name in the lock-screen would solve the problem.

benma, 2021-11-03:
Here is a proposal PR: https://github.com/digitalbitbox/bitbox02-firmware/pull/826 - we will go through some testing and decide if we can include this in the next firmware release.

benma, 2024-06-06:
The PR above is now finally merged, so this feature will be available with the next firmware release.

closed 2024-06-06 by benma