BitBoxSwiss/bitbox02-firmware #842: Increase the timeout time when inputing password.

alexwaltz, 2021-11-29:
I generated a seed with the provided dicewear.

After that, I restored a wallet with the freshly generated seed.
When this step was over I was asked to input a password.

As I wanted a safe password I generated it on another device.
This resulted in a password that I can't remember, so I need to move my head around as I input the password. 
After a certain number of characters are entered I lose track of where I am and I will need to count the characters, as I am doing this the BitBox times out and I have to start the process from 0.

Even though I imagine this is a security feature, I think the time-out is too short, and makes for a very frustrating experience.

benma, 2021-11-30:
There is actually no timeout when entering a password. Do you use an adapter or a cable? It could be that these are faulty or that the connection is otherwise unstable (loose contact).

If you use an adapter or a cable, try connecting directly without, or with another adapter.

benma, 2021-11-30:
@alexwaltz No adaptor, also no cable?

Could there be dust or other things in your USB slot that could be cleaned out?

Please contact support@shiftcrypto.ch, if the device is faulty it will be replaced.

alexwaltz, 2021-11-30:
I do want to clarify that in the end, it worked, and I did not have any problems since then.

The issue was just at the specific point I mentioned.

I will clean up and restore the seed once again, and let you know if the problem persists.
And if the issue persists, I will also try another device.

I will report back when done.

benma, 2021-11-30:
@alexwaltz I see, good to hear. Would be great to know if there is actually some problem that only appears when entering the password.

If it does, you can also try confirming an address on the device and leaving it on the screen without dismissing it for a while. It should never disappear unless dismissed by touch.

benma, 2022-01-24:
Closing this as it appears to be solved. Please re-open otherwise.

closed 2022-01-24 by benma