BitBoxSwiss/bitbox02-firmware #843: Make Opensea transactions work

benma, 2021-11-29:
The size of the contract data in Opensea transactions is larger than the BitBox02 accepts, leading to an 'Invalid Input' error.

The size limit should be increased to accommodate opensea transactions.

Ideally, Opensea transactions would be parsed and displayed in a human readable format instead. The current way to display the raw contract data is only safe for expert users that know how to verify it, but that is not feasible for huge contract data fields like with Opensea.

EVLRBT-01, 2021-11-29:
Thanks for this. Would EIP-712 support help in parsing the contract data?

Discussions at https://github.com/trezor/trezor-firmware/pull/1568 and https://github.com/LedgerHQ/app-ethereum/issues/105 might provide some useful context.

benma, 2021-11-30:
EIP712 seems to be a message signing format. I am not sure if Opensea uses this, but this issue is about parsing the contract data field in a transaction to the opensea contract, which is onchain - message signing is independent of that I think.

EVLRBT-01, 2021-12-01:
Hmmm, does the whole contract get passed to the device for signing? My naive thought was that it sent the message for signing, which has a JSON payload of all the relevant info to be shown on the device.

Perhaps not the issue as you suggest, but may be useful nonetheless.

benma, 2021-12-01:
For an onchain transaction, the contract data is inside the transaction, and the whole transaction is sent to the device for signing. This would happen e.g. when you buy an NFT I think.

There are other operations that require message signing (EIP712), e.g. listing a new NFT.

remixie, 2021-12-02:
@x1ddos @benma  Don't set this ticket as an "enhancement". There is a lot of people right now mad that their trezor wallets have Opensea NFTs held hostage due to lack of support. Ledger already has this issue resolved a while ago. Get this fixed. This is not a request. Due to the time sensitive nature of tons of my NFT's, my portfolio is going to take a huge loss and there will be consequences if this happens.

remixie, 2021-12-02:
I deleted my previously heated message. I would like however for this to be fixed in a timely manner because I am genuinely afraid that my portfolio will take a huge loss due to the time-sensitive nature of NFTs at the moment.

x1ddos, 2021-12-02:
so, what's the max size of contract data, is it spec'ed anywhere? does [EIP-170](https://github.com/ethereum/EIPs/blob/master/EIPS/eip-170.md) relate to this at all?

EVLRBT-01, 2021-12-02:
I'm not sure if this should work, but tried a similar action on the Polygon network using Rabby and got as far as submitting a transaction to unlock selling via OpenSea. The sign transaction dialog pops up, I click Sign and then nothing appears on the BB02 (no errors on console). 

Having said that, trying to send MATIC on Polygon had same result, which is surprising as it is an ERC20 token. I figured it would just be the difference in testnet/mainnet.

benma, 2022-01-10:
@EVLRBT-01 the polygon network is not supported by the BitBox02.

PR #858 (2022-01-10): eth/sign: increase size limit for tx data
Fixes #843

We previously limited the max. size of contract data to 1024 bytes.

Some smart contract invocations require much larger data,
e.g. transactions involving the Opensea contract. Example tx with a
large data field (~2400 bytes):

https://etherscan.io/tx/0x2db260bf407a246e3bbf57692168238b55b8518ef72c8288e68547e418cf0e08

Ideally, Opensea transactions would be parsed and displayed in a human
readable format instead. The current way to display the raw contract
data is only safe for expert users that know how to verify it, but
that is not feasibl [... cut]
closed 2022-01-10 by benma