BitBoxSwiss/bitbox02-firmware #862: Feature: Hitting back during "keypad" input goes back to previous group of characters

hojdan, 2022-01-15:
Let's say I want to enter the letter a as a simple example.

1. I touch the group with a-h
2. I touch the next group that appears which is a-c
3. I then touch a

Let's say I click something wrong by accident at some point.

1. If I touch the next group i-q, touching the back button afterward will take me back to all the letters and then I can touch the correct group. No issue here.
2. If I touch the next group d-e, touching the back button afterward will take me all the way back to all the letters. Then I would need to first touch group a-h from all the letters then I can touch the correct group. I would prefer that touching the back button not take me back all the way but instead to just the previous group selection.
3. I click b by accident and thus b is entered. To correct my mistake I click the back button. No issue here.

benma, 2022-01-21:
Makes sense to me. @jadzeidan what do you think?

jadzeidan, 2022-01-21:
Yeah, also makes sense to me to go back to previous group. Ack from me.