BitBoxSwiss/bitbox02-firmware #864: Error building firmware Dockerimage

cstenglein, 2022-01-22:
Tried to build bitcoin-only firmware 9.9.0 with the build.sh script:

`./build.sh firmware-btc-only/v4.1.0 "make firmware-btc"`

Error log: 

```bash
Step 11/21 : RUN pip3 install     hidapi     noiseprotocol     protobuf     ecdsa     tzlocal     semver
 ---> Running in efe399bf61a8
Collecting hidapi
  Downloading https://files.pythonhosted.org/packages/dc/aa/38708a1d85d13dec22e756feb4e02f8b3adc5937bfe976f8f998717ff0a3/hidapi-0.11.0.post2.tar.gz (136kB)
Collecting noiseprotocol
  Downloading https://files.pythonhosted.org/packages/9d/e1/76e4694201d67b93a6f1644b2588b4a3d965419fe189416e3496cf415db5/noiseprotocol-0.3.1-py3-none-any.whl
Collecting protobuf
  Downloading https://files.pythonhosted.org/packages/dc/a9/fdc31f2c7917d5a1e18f4ca2f955706dfdcaaba8fb0baa48becfc0b68f22/protobuf-3.19.3-py2.py3-none-any.whl (162kB)
Collecting ecdsa
  Downloading https://files.pythonhosted.org/packages/4a/b6/b678b080967b2696e9a201c096dc076ad756fb35c87dca4e1d1a13496ff7/ecdsa-0.17.0-py2.py3-none-any.whl (119kB)
Collecting tzlocal
  Downloading https://files.pythonhosted.org/packages/ac/29/0daef8da885f35b3c8a54ac45378d4829a7f787ec55475b541bed3df6a36/tzlocal-4.1-py3-none-any.whl
Collecting semver
  Downloading https://files.pythonhosted.org/packages/0b/70/b84f9944a03964a88031ef6ac219b6c91e8ba2f373362329d8770ef36f02/semver-2.13.0-py2.py3-none-any.whl
Requirement already satisfied: setuptools>=19.0 in /usr/lib/python3/dist-packages (from hidapi)
Collecting cryptography>=2.8 (from noiseprotocol)
  Downloading https://files.pythonhosted.org/packages/f9/4b/1cf8e281f7ae4046a59e5e39dd7471d46db9f61bb564fddbff9084c4334f/cryptography-36.0.1.tar.gz (572kB)
    Complete output from command python setup.py egg_info:
    
            =============================DEBUG ASSISTANCE==========================
            If you are seeing an error here please try the following to
            successfully install cryptography:
    
            Upgrade to the latest pip and try again. This will fix errors for most
            users. See: https://pip.pypa.io/en/stable/installing/#upgrading-pip
            =============================DEBUG ASSISTANCE==========================
    
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-5kivx15e/cryptography/setup.py", line 14, in <module>
        from setuptools_rust import RustExtension
    ModuleNotFoundError: No module named 'setuptools_rust'
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-5kivx15e/cryptography/
Removing intermediate container efe399bf61a8
The command '/bin/sh -c pip3 install     hidapi     noiseprotocol     protobuf     ecdsa     tzlocal     semver' returned a non-zero code: 1
```

benma, 2022-01-24:
> ./build.sh firmware-btc-only/v4.1.0 "make firmware-btc"

Could it be because the version is wrong? 9.9.0 vs 4.1.0.

cstenglein, 2022-01-24:
Yeah, you are right, my bad :O

=> Closed

closed 2022-01-24 by cstenglein