BitBoxSwiss/bitbox02-firmware #880: Support for chainId greater than 8 bits

skenaja, 2022-02-21:
It looks like the firmware expects the chainId to be 8 bits in length for signing transactions, i.e. max value 127. This prevents support for EVM networks like Palm Network which has a chainId = 11297108109 (so 64 bits).  There are many other networks with higher chainIds at https://chainlist.org/ including Polygon & Avalanche.

Is this a hardware limitation, or could it be supported?

benma, 2022-02-21:
There is no hardware limitation, but we only support the ETH chain and Rinkeby/Ropsten testnet chains at the moment. cc @jadzeidan

benma, 2022-02-25:
Support for chain ids up to 8 bytes: https://github.com/digitalbitbox/bitbox02-firmware/pull/881/

benma, 2022-03-10:
v9.10.0 will support Binance Smart Chain, Optimism, Polygon, Fantom Opera and Arbitrum One networks.

closed 2022-03-10 by benma