BitBoxSwiss/bitbox02-firmware #885: Did I get a dud?

Transisto, 2022-03-03:
I brought this last year at 2021 conference,

https://www.youtube.com/watch?v=iT0AHXrKig0

I'm not interested in testing this wallet further.

So far the process was straightforward, Is there a version with physical buttons? I don't think I can stand the zero-feedback and spotty responsiveness.

Transisto, 2022-03-03:
BTW, I'm trying to type the password TEST

benma, 2022-03-03:
Hi

It could be a defective device. Please contact support@shiftcrypto.ch to get it sorted out.

These tips might also help: https://shiftcrypto.support/help/en-us/9-troubleshooting/16-touch-sensor-issues 

>  Is there a version with physical buttons

I'm afraid there isn't. The sides are not just touch buttons but touch sliders, letting you e.g. slide through an address or a info text.

closed 2022-03-03 by benma