BitBoxSwiss/bitbox02-firmware #910: Support for Well-Known Blockchains

SiNONiMiTY, 2022-05-05:
BitBox02 Specifications are very good, one thing that hinders its adoption alongside Trezor and Ledger is the Blockchain Support.

Blockchains like
- Algorand (ALGO)
- Tezos (XTZ)
- PolkaDot (DOT)
- Stellar (XLM)
- Monero (XMR)
- VeChain (VET)
- Ergo (ERG)

Personally, I would like to avail one, but because of the limited blockchain support, I am leaning more towards to Ledger.

ShiftCrypto Team, your device is absolutely amazing, but please help integrate other chains outside of the EVM compatibility layer.

jadzeidan, 2022-05-05:
@SiNONiMiTY Thanks for the feedback and happy to hear you like the BitBox02 🙂 

We tend to be conservative when adding more altcoins because it's a lot of effort to maintain and the major coins can change overtime. However if we get consistent requests for a particular coin and we see it has a strong community that is likely to be around in the long run, then we could consider adding it.

SiNONiMiTY, 2022-05-05:
> 

Hi @jadzeidan 

I think adding the chains proactively would be better. Some of the chains I mentioned here are among the top 100 and some have been around for too long. BitBox02 seems really good but if it doesn't even support what is frequently used by the users, then they would just have to find something that works for them, leaving out BitBox02's potential.

I hope the team considers adding chains little by little. Show the users how awesome your hardware is.

Messa1, 2022-05-23:
There are so many other coins that need to be added. I like your hardware wallet. But other support more coins.

satoshinotdead, 2023-03-12:
Monero is a true crypto-currency, not the rest.

benma, 2023-05-29:
We are currently not planning on adding support for other coins - we don't see consistent and large demand for any of them at the moment. If we notice increased demand for a particular coin we will consider adding it.

closed 2023-05-29 by benma

satoshinotdead, 2023-08-23:
Just to reminder that Monero is the only true privacy coin out there.

I'm afraid we would be forced to stop using and promoting Bitbox until you give us support.

dylangerdaly, 2023-12-21:
+1 for Monero (XMR) Support, that'd be a massive gain towards uptake

satoshinotdead, 2023-12-24:
> +1 for Monero (XMR) Support, that'd be a massive gain towards uptake

That's where the real cypherpunk fundamentals. I sold my Bitbox because of this total lack of support.

greatwolf, 2024-11-09:
just wondering, would it be possible to fork the firmware modify it to support your coin and then load it onto your bitbox device?