BitBoxSwiss/bitbox02-firmware #911: Output Script Descriptor (with Miniscript) support

darosior, 2022-05-06:
[Output Script Descriptor](https://github.com/bitcoin/bips/blob/master/bip-0380.mediawiki#motivation) are a sane standard for wallet backups. They express all the information necessary to locate owned (or partially owned) outputs in an engineer-readable format, avoiding the pitfalls of previously used implicit hacks to recover only from a mnemonic.

[Miniscript](https://bitcoin.sipa.be/miniscript/) is an extension to Output Script Descriptors, which defines a language to reason about a subset of Script. Miniscript allows to:
1. Safely write more complicated contracts than the basic well-known templates
2. Produce a valid witness for any Script that is a valid Miniscript, given the necessary material (signature, etc)
3. Statically analyse the properties of any Script that is a valid Miniscript: determine the semantics of the contract, whether it is non-malleable, whether some spending paths may exceed standardness or consensus limits --in short, whether it's safe to participate in such a Script.

Miniscript also guarantees consensus soundness (unless the conditions of the Miniscript are met, no witness can be created for the Script) and standardness completeness (for any Miniscript that was analysed as sane, a witness can be constructed in the bounds of the consensus and standardness rules).

Output Script Descriptors along with Miniscript are particularly appealing to signing devices, as it permits the user of this device to safely take part in more complex contracts. It also gives a framework to add support for new Scripts, instead of relying on hard-coded Script templates and footgun-y backup solutions.

Output descriptors have been in Bitcoin Core for a while now, and [Miniscript was merged into Bitcoin Core](https://github.com/bitcoin/bitcoin/pull/24147) mainline this year. The [Specter](https://github.com/cryptoadvance/specter-diy) signing device added support for Output descriptors and Miniscript more than a year ago. [Ledger added support](https://blog.ledger.com/bitcoin-2/) for (a subset of) output descriptors last year and is [currently working toward Miniscript](https://blog.ledger.com/miniscript-is-coming/) support.
Is there any plan to implement to implement Output Script Descriptors (with Miniscript) for the Bitbox2?

darosior, 2022-05-06:
Note that [libwally is implementing Miniscript and Output Descriptors](https://github.com/ElementsProject/libwally-core/pull/310), since it is already a dependency of this project it would very likely not be necessary to re-implement it here.

benma, 2022-05-13:
Hi

Do you know of any user wallets where descriptors with miniscript can be imported and used?

Are there compelling use cases for end users?

I'd love to add support for it, but we are stretched thin demand for this seems very low at the moment.

papawcoin, 2023-06-25:
I want to fork

benma, 2023-10-10:
The Rust library has been published a while ago: https://crates.io/bitbox-api

Miniscript support was released in [v9.15.0](https://github.com/digitalbitbox/bitbox02-firmware/releases/tag/firmware%2Fv9.15.0).

closed 2023-10-10 by benma