BitBoxSwiss/bitbox02-firmware #914: OpenSea Listing: EIP-712

EVLRBT-01, 2022-05-13:
Good afternoon,

It seems that I can no longer list items on OpenSea, probably due to the fact that transactions are now 'typed' as per [EIP-712](https://eips.ethereum.org/EIPS/eip-712). Some information at https://www.nft-stats.com/info/technical-analysis-open-sea-phishing-attack describes this change. I did experience the issue several weeks ago and hadn't tried again until now, so it does seem directly related to the contract change.

As an example, when I try to list a new item it first prompts for the normal OS approval to sell on your behalf and the Rabby UI states "Sign Ethereum transaction" which works fine. Once the approval transaction completes, it then prompts for the listing transaction, which the Rabby UI states as "Sign Typed Message" along with some JSON describing the transaction data. Submitting this results in the following in the browser console and no prompt on the BB02:

[image]

I did also previously mention this at https://github.com/digitalbitbox/bitbox02-firmware/issues/843#issuecomment-982066833.

BB02 9.10.0
Rabby 0.30.1

benma, 2022-05-13:
Thanks for the report. Is there a legacy/compatibility mode to do it without eip712 messages as before?

EVLRBT-01, 2022-05-13:
I'm not sure, but I'm guessing that due to the new smart contract as of a few months ago, that is now the standard way of doing it to better protect users. 

LooksRare, another NFT platform also seems to use EIP-712 but I havent tried there yet. They at least have more accessible documentation - https://docs.looksrare.org/developers/maker-orders. 

The JSON on the OS listing also seems to be a 'maker order', with the following output (scrubbed) from the Rabby popup:

`{
    "maker": "",
    "exchange": "",
    "taker": "",
    "makerRelayerFee": "",
    "takerRelayerFee": "",
    "makerProtocolFee": "",
    "takerProtocolFee": "",
    "feeRecipient": "",
    "feeMethod": 0,
    "side": 0,
    "saleKind": 0,
    "target": "",
    "howToCall": 0,
    "calldata": "",
    "replacementPattern": "",
    "staticTarget": "",
    "staticExtradata": "",
    "paymentToken": "",
    "basePrice": "",
    "extra": "0",
    "listingTime": "",
    "expirationTime": "",
    "salt": "",
    "nonce": 0
}`

OS uses the Wyvern Protocol I believe and it seems this came into effect with [Wyvern 2,3](https://github.com/ProjectOpenSea/opensea-js/pull/362) in February. It seems that a [Wyvern v3](https://wyvernprotocol.com/docs) may be on the horizon but cant tell if that will have any additional impact.

EVLRBT-01, 2022-05-13:
I also had a problem with BB02 on snapshot.org in the past, also probably relating to EIP-712 support. There is a tracking issue at https://github.com/snapshot-labs/snapshot/issues/950 for general support: seems typed data array support might be something else to look out for.

EVLRBT-01, 2022-05-22:
Another development: "Every Seaport listing consists of the same basic structure, including an improved EIP-712 signature payload that clearly outlines what can be spent and what will be received back by whom."

https://opensea.io/blog/announcements/introducing-seaport-protocol

benma, 2022-05-27:
Work on this has started.

EVLRBT-01, 2022-05-27:
Excellent, great to hear!

EVLRBT-01, 2022-06-14:
FYI - OpenSea has now moved to using SeaPort Protocol.

twitter.com/opensea/status/1536756396158599168

benma, 2022-06-14:
We are releasing support for EIP-712 soon. The PR implementing it was merged (https://github.com/digitalbitbox/bitbox02-firmware/pull/920), v9.12.0 is [tagged](https://github.com/digitalbitbox/bitbox02-firmware/blob/firmware/v9.12.0/CHANGELOG.md#9120), and Rabby is [ready](https://github.com/RabbyHub/Rabby/releases/tag/v0.34.0).

benma, 2022-08-21:
v9.12.0 was released the same day, but I forgot to update this issue. Apologies.

closed 2022-08-21 by benma