BitBoxSwiss/bitbox02-firmware #939: Firmware upgrade without the app

real-or-random, 2022-07-01:
Is there a (reasonable) way to upgrade the firmware without using the app?

benma, 2022-07-02:
You can use this Python script: https://github.com/digitalbitbox/bitbox02-firmware/blob/master/py/load_firmware.py after installing the bitbox02 package here: https://github.com/digitalbitbox/bitbox02-firmware/tree/master/py (or install it directly from PyPi).

Example:

    ./py/load_firmware.py firmware-btc.v9.12.0.signed.bin

real-or-random, 2022-07-02:
Thanks! It may be good to document that somewhere, I couldn't find that piece of information.

benma, 2022-07-03:
:+1: does this look good? https://github.com/digitalbitbox/bitbox02-firmware/pull/940

closed 2022-07-04 by benma

real-or-random, 2022-07-14:
I tried this now. Here's a bug:

* `> python load_firmware.py firmware-btc.v9.12.0.signed.bin`
* enter PIN on bitbox
* say yes on bitbox
* bootloader is loaded on bitbox
* output: 
 ```
.{'interface_number': 0,
 'manufacturer_string': 'shiftcrypto.ch',
 'path': b'/dev/hidraw8',
 'product_id': 9219,
 'product_string': 'bb02btc-bootloader',
 'release_number': 256,
 'serial_number': 'v1.0.3',
 'usage': 1,
 'usage_page': 65535,
 'vendor_id': 1003}
Traceback (most recent call last):
  File "/tmp/bitbox/load_firmware.py", line 165, in <module>
    sys.exit(main())
  File "/tmp/bitbox/load_firmware.py", line 136, in main
    bootloader_device, transport = _find_and_open_usb_bitbox02(not args.no_cache)
  File "/tmp/bitbox/load_firmware.py", line 116, in _find_and_open_usb_bitbox02
    hid_device.open_path(bootloader_device["path"])
  File "hidraw.pyx", line 142, in hid.device.open_path
OSError: open failed
```
* run `> python load_firmware.py firmware-btc.v9.12.0.signed.bin` again
* update works.

This is reproducible. I guess this is some race condition where we try to open the device before it's ready.

----

One more question. The docs say: 
> an option can be turned on to display the firmware hash on the device screen when the BitBox02 is plugged in. 

But how to turn it on? Sorry I couldn't figure it out...

benma, 2022-07-15:
Is maybe the BitBoxApp open or some other wallet that could be racing for a connection to the device?

> But how to turn it on? Sorry I couldn't figure it out...

With the BitBoxApp, go to Manage device -> go to startup settings -> see the toggle.

Without the app: use `py/send_message.py` to reboot into the bootloader, then again `py/send_message.py` to see some bootloader commands. One of them should be to toggle this.

real-or-random, 2022-07-15:
> Is maybe the BitBoxApp open or some other wallet that could be racing for a connection to the device?

Hm, I don't think so but I'll retest.

benma, 2022-07-15:
Does it happen every time? That would indicate a bug indeed. Never saw it before though unless some other wallet grabs the connection, but that is a bit unlikely seeing that you already were connected before the reboot.

As a potential workaround you can just re-run the script while the BitBox02 is in bootloader mode and seeing if that helps.

Does the BitBoxApp work normally?

Fyi, the send_message.py tool is more of a messy dev tool than a proper CLI client.

real-or-random, 2022-07-15:
> As a potential workaround you can just re-run the script while the BitBox02 is in bootloader mode and seeing if that helps.

Yeah that worked, see above.

> Does the BitBoxApp work normally?

Yes.