BitBoxSwiss/bitbox02-firmware #949: Problem connecting to Sparrow

santochibtc, 2022-07-10:
Hi,
I can't connect the device with Sparrow Desktop on Windows 10. Nothing happens when I press the Scan button in the "connect hardware wallet" window. The device always shows "See the BitBoxApp". If I open the BitBox app, unlock the device with the PIN and close the app, then Sparrow shows "Error parsing JSON" after pressing the Scan button.

benma, 2022-07-10:
Hi

Which version of the BitBox02 firmware and of Sparrow are you running?

We tried to reproduce the problem but it seems to work as expected.

santochibtc, 2022-07-10:
Sparrow 1.6.5 and BitBox02 firmware 9.12.0

benma, 2022-07-10:
I am not so familiar with Sparrow. Maybe you can find some log files and see if they reveal anything.

Maybe you can ask the Sparrow developers, "Error parsing JSON" seems like it might be a bug in Sparrow.

santochibtc, 2022-07-10:
I found this issue with Trezor
https://github.com/sparrowwallet/sparrow/issues/542
that led me to this one in the HWI library, but I am not sure if it is the same problem
https://github.com/bitcoin-core/HWI/issues/613

santochibtc, 2022-07-10:
I am trying to debug the enumerate function in the HWI library and it crashes in bitbox02.py while doing:
from bitbox02 import util
error:
Descriptors cannot not be created directly.
If this call came from a _pb2.py file, your generated code is out of date and must be regenerated with protoc >= 3.19.0.
If you cannot immediately regenerate your protos, some other possible workarounds are:
 1. Downgrade the protobuf package to 3.20.x or lower.
 2. Set PROTOCOL_BUFFERS_PYTHON_IMPLEMENTATION=python (but this will use pure-Python parsing and will be much slower).
But I think this is a problem with my Python setup and it is not related with the problem in Sparrow

benma, 2022-07-11:
Thanks a lot for debugging. Upgrading protoc is actually on our radar. CC @Beerosagos 

In the meantime you can downgrade your local installation of protobuf I think.

santochibtc, 2022-07-11:
I have downgraded protobuf to 3.20. Now the enumerate method returns a correct string with the device
[{"type": "bitbox02", "path":..."fingerprint": "XXXX"}]
I have removed jade from the list of devices, the method never ends otherwise.
I will see if I can now embed this HWI library in Sparrow

zerotobtc, 2022-07-22:
I have a similar issue on Arch Linux. I get the same error when I try to create a new wallet in Sparrow from the bitbox02. The error log looks like this:
` JSON: 
com.sparrowwallet.sparrow.io.ImportException: Error scanning, check devices are ready
	at com.sparrowwallet.sparrow@1.6.5/com.sparrowwallet.sparrow.io.Hwi.enumerate(Unknown Source)
	at com.sparrowwallet.sparrow@1.6.5/com.sparrowwallet.sparrow.io.Hwi$EnumerateService$1.call(Unknown Source)
	at com.sparrowwallet.sparrow@1.6.5/com.sparrowwallet.sparrow.io.Hwi$EnumerateService$1.call(Unknown Source)
	at javafx.graphics@18/javafx.concurrent.Task$TaskCallable.call(Unknown Source)
	at java.base/java.util.concurrent.FutureTask.run(Unknown Source)
	at javafx.graphics@18/javafx.concurrent.Service.lambda$executeTask$6(Unknown Source)
	at java.base/java.security.AccessController.doPrivileged(Unknown Source)
	at javafx.graphics@18/javafx.concurrent.Service.lambda$executeTask$7(Unknown Source)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.base/java.lang.Thread.run(Unknown `Source)`

I was able to do that back in 2021, but I remember having trouble there too. Any suggestions?
Sparrow: 1.6.5

benma, 2022-07-22:
@zerotobtc have you tried downgrading protobuf as described above?

`pip3 install protobuf==3.20`.

We are preparing a PR to HWI that fixes the protobuf incompatibility, but it may take a while until it trickles down to Sparrow.

benma, 2023-05-29:
Closing due to inactivity. Please comment if the issue persists. Thanks.

closed 2023-05-29 by benma