BitBoxSwiss/bitbox02-firmware #995: Double password input speed by using both sides

ln2max, 2022-10-22:
The password input process currently only uses one side of the device. As a result, at each stage of the search tree, the user can choose from only 3 possibilities.

However, the Bitbox02 has touch sensors on both sides of the device. By using the other sensors as well, the user would be presented with 6 choices at each step of the input tree. This would reduce the time to reach the root of the tree to approximately half, dramatically speeding up the somewhat cumbersome password input process.

jadzeidan, 2022-10-24:
Hey @ln2max, thanks for the input. The password input process is a balance between increasing complexity to acheive speed and vice versa. As an advanced user what you described might be more efficient, but for other users that aren't as advanced or aren't as dexterous it may make the process harder. However if we find time to improve the password entry one day, this proposal is something we could experiment with.

benma, 2023-05-29:
Closing this as it is unlikely we will go this way. Our newest experiments change the way the keyboard is switched but the top bar is still needed for it.

closed 2023-05-29 by benma