BitBoxSwiss/bitbox02-firmware #996: Can't reproduce builds

caheredia, 2022-10-24:
I'm following the directions from the [releases](https://github.com/digitalbitbox/bitbox02-firmware/tree/master/releases) page. But the build is failing. Perhaps I'm doing something wrong? 

```console
git clone --progress https://github.com/digitalbitbox/bitbox02-firmware
cd bitbox02-firmware/releases 
git checkout firmware-btc-only/v9.12.0 
./build.sh firmware-btc-only/v9.12.0 "make firmware-btc"
```

The script kicks off the build process, but eventually it fails to locate a dependency
```shell
[+] Building 53.3s (9/38)                                                                                                                                  
 => [internal] load build definition from Dockerfile                                                                                                  0.0s
 => => transferring dockerfile: 4.75kB                                                                                                                0.0s
 => [internal] load .dockerignore                                                                                                                     0.0s
 => => transferring context: 74B                                                                                                                      0.0s
 => [internal] load metadata for docker.io/library/ubuntu:18.04                                                                                       0.6s
 => [internal] load build context                                                                                                                     0.1s
 => => transferring context: 481.73kB                                                                                                                 0.0s
 => CACHED [ 1/34] FROM docker.io/library/ubuntu:18.04@sha256:40b84b75884ff39e4cac4bf62cb9678227b1fbf9dbe3f67ef2a6b073aa4bb529                        0.0s
 => [ 2/34] RUN apt-get update && apt-get upgrade -y && apt-get install -y wget nano rsync curl gnupg2 jq                                            14.2s
 => [ 3/34] RUN echo "deb http://apt.llvm.org/bionic/ llvm-toolchain-bionic-8 main" >> /etc/apt/sources.list &&     echo "deb-src http://apt.llvm.or  0.5s 
 => [ 4/34] RUN mkdir ~/Downloads &&    cd ~/Downloads &&    wget -O gcc.tar.bz2 https://developer.arm.com/-/media/Files/downloads/gnu-rm/8-2018q4/  35.7s 
 => ERROR [ 5/34] RUN apt-get update && apt-get install -y     build-essential     llvm     gcc-8     binutils     valgrind     cmake     git     au  2.2s 
------                                                                                                                                                     
 > [ 5/34] RUN apt-get update && apt-get install -y     build-essential     llvm     gcc-8     binutils     valgrind     cmake     git     autotools-dev   
  automake     autoconf     libtool     pkg-config     libcmocka-dev     libc6-i386     lib32stdc++6     lib32z1     libusb-1.0-0-dev     libudev-dev     l
ibhidapi-dev:                                                                                                                                              
#8 0.326 Get:1 https://apt.llvm.org/bionic llvm-toolchain-bionic-8 InRelease [4231 B]                                                                      
#8 0.404 Get:2 https://apt.llvm.org/bionic llvm-toolchain-bionic-8/main Sources [2548 B]                                                                   
#8 0.564 Hit:3 http://ports.ubuntu.com/ubuntu-ports bionic InRelease
#8 0.700 Hit:4 http://ports.ubuntu.com/ubuntu-ports bionic-updates InRelease
#8 0.839 Hit:5 http://ports.ubuntu.com/ubuntu-ports bionic-backports InRelease
#8 0.977 Hit:6 http://ports.ubuntu.com/ubuntu-ports bionic-security InRelease
#8 1.052 Fetched 6779 B in 1s (8808 B/s)
#8 1.052 Reading package lists...
#8 1.567 Reading package lists...
#8 2.046 Building dependency tree...
#8 2.143 Reading state information...
#8 2.179 E: Unable to locate package libc6-i386
#8 2.179 E: Unable to locate package lib32stdc++6
#8 2.179 E: Couldn't find any package by regex 'lib32stdc++6'
#8 2.179 E: Unable to locate package lib32z1
------
executor failed running [/bin/sh -c apt-get update && apt-get install -y     build-essential     llvm     gcc-8     binutils     valgrind     cmake     git
     autotools-dev     automake     autoconf     libtool     pkg-config     libcmocka-dev     libc6-i386     lib32stdc++6     lib32z1     libusb-1.0-0-dev 
    libudev-dev     libhidapi-dev]: exit code: 100
```

benma, 2022-10-24:
Hi @caheredia, thanks for trying to reproduce the firmware.

I just tried it and it worked successfully. I think this kind of message can appear if you have network problems. Please check your connection, and also restart Docker (`sudo service docker restart`), as the network in Docker sometimes breaks in my experience.

benma, 2022-10-24:
Fyi, this line is not needed:

> git checkout firmware-btc-only/v9.12.0 

In this instance, I don't think it makes a difference, but in some cases it could. It is best to use the most up-to-date build script (which takes care of checking out the specified version).

benma, 2022-11-05:
@caheredia did you find the time to try again?

caheredia, 2022-11-05:
@benma , I think the issue might be that I'm running the `build.sh` script on my  on mac (M1). I'll try later tonight after forcing the Docker container to be x86 architecture with `--platform linux/amd64`

caheredia, 2022-11-07:
got it to work on an Intel mac. Thanks for following up
```console
shasum -a 256 bitbox02-firmware/releases/temp/build/bin/firmware-btc.bin
1d3cf66e1b2f91e369a434ee81700323896c7e8c51a79a20de13a3c90099cebd  bitbox02-firmware/releases/temp/build/bin/firmware-btc.bin
```

closed 2022-11-07 by caheredia

benma, 2022-11-08:
@caheredia was `--platform linux/amd64` necessary in the end? Or was it something else?