BitBoxSwiss/bitbox02-firmware #1047: Include a secure method to delete SD Card

satoshinotdead, 2023-03-11:


benma, 2023-05-26:
Closing as `wontfix` according to the above comment.

closed 2023-05-26 by benma

satoshinotdead, 2023-05-28:
> Hi.
> 
> The sdcard backup is equivalent to the 24 recovery words that you can write down on paper, which are also not encrypted.
> 
> We didn't include a way to delete the backup because it is not clear if/how the bytes can really be wiped securely, and also to avoid users accidentally deleting important backups. Better to have one too many backups than one too few.
> 
> Hope this helps.

That's not equivalent to 24-word because I already got the 24-word.

That's an obligated huge security risk that you software need to make an option or solve adding a secure way to destroying it.

benma, 2023-05-29:
> That's not equivalent to 24-words because I already got the 24-words.

I meant equivalent with respect to encryption: your 24 words are not encrypted, the sdcard backup is not encrypted. You can store both in the same location with the same security.

benma, 2023-05-29:
By the way, we plan on allowing wallet creation without making an sdcard backup at all, which could also alleviate your concern.

satoshinotdead, 2023-05-29:
> I meant equivalent with respect to encryption: your 24 words are not encrypted, the sdcard backup is not encrypted.

Sorry but I'm not talking about my written seed and you really don't know if I have it encrypted or not.

> we plan on allowing wallet creation without making an sdcard backup at all

Fair enough, that will solve the issue. May I ask an _approximate_ ETA? Next big update, probably?

Thanks for your following up 👍

benma, 2023-05-29:
>  May I ask an approximate ETA? Next big update, probably?

It is planned and in the works, that means it will likely be in the next release or in the release after that (though note that priorities can sometimes change on short notice).