BitBoxSwiss/bitbox02-firmware #1107: Anti-klepto for Ethereum and Cardano?

jcw1tjjrdhv3, 2023-08-06:
Does anti-klepto work with Ethereum (via MyEtherWallet) and Cardano? Sorry if this is a wrong place to ask, but I couldn't find an answer anywhere on the website. Perhaps this is something that could be added to the knowledgebase.

benma, 2023-08-06:
It is supported for all ECDSA signatures, which means it is supported for Ethereum, also via MEW. Cardano does not use ECDSA, so there is no support for anti-klepto in Cardano. See also https://bitbox.swiss/bitbox02/threat-model/.

closed 2023-08-06 by benma

jcw1tjjrdhv3, 2023-08-06:
I see, thank you. But can it also be implemented for Cardano and if yes, are there any plans to do so in the future?

benma, 2023-08-06:
We haven't looked into if it's possible for Cardano. There are no plans for that.

jcw1tjjrdhv3, 2023-08-06:
OK, thanks for the info.