BitBoxSwiss/bitbox02-firmware #1125: BitBox fails to sign segwit multisig following firmware update

dr-orlovsky, 2023-09-07:
Bitbox fails to sign simple n-of-n segwit (wsh) multisig PSBT input, not recognizing its fingerprint/derivation path:

`{"error": "This BitBox02 is not one of the cosigners", "code": -7}`

The derivation path is `362a0411/48'/0'/2'/2'` and is present in PSBT in correct form, including fingerprint matching the device fingerprint.

The wallet was not registered on BitBox device previously - but with previous version of firmware it was asking about registering new wallet in such siguation

benma, 2023-09-07:
Nothing has changed in the BitBox02 multisig implementation, so it is likely that the problem is with your PSBT.

Your PSBT must contain all cosigner xpubs in the global xpubs field of the PSBT, and one of the xpubs must be the BitBox02's xpub at the given derivation path and have the correct fingerprint. 

Probably either the xpub is missing from your PSBT, or your BitBox02 has a different seed loaded so the BB02 xpub does not match any of the ones given in the PSBT.

If the issue persists, please provide a full PSBT and a BitBox02 seedphrase to reproduce the problem.

dr-orlovsky, 2023-09-07:
That was a weird tx - other simple multisig tx are got signed. It's a mainnet tx so I can't share a PSBT, however I can't see any issues with it. Anyway, seems to be a false report and I just do not see what's wrong with that PSBT. Closing

closed 2023-09-07 by dr-orlovsky