BitBoxSwiss/bitbox02-firmware #1126: Not able to sign timelocked PSBT via HWI without wallet registration

dr-orlovsky, 2023-09-07:
I know it is a feature and not a bug - but maybe there is some way to solve that, since it works for non-timelocked multisigs.

What I get is 
```
BitBox02 attestation check PASSED
{"error": "Input or change script type not recognized.", "code": -7}
```

Descriptor is 
```
wsh(or_i(
    multi(3,
            [362a0411/48h/0h/0h/2h]xpub6DnLtr7gtXNFZJcDLni2ezMrtaQRLDQTvW8rdyWn33pWUymmeocvNTaQyBZWEYPosEJehmfsbe2A8Tg9jS2YwKWjiwAkuRZ6vzVjFs52Juf/<0;1>/*,
            [dbd96ebc/48h/0h/0h/2h]xpub6F8HEj83d5th3okdS2eJ4r1eNmcHw7VtLUzqnMRq6KNr7bajS8hKUUUzxjAvRMH5jdbzpR3qfBRhKfsTpBYFu6Pba1UjKLZN95sB8DEo3Ja/<0;1>/*,
            [feda4d11/48h/0h/0h/2h]xpub6EgHtjXXbGmVHEuKw49fke3iqXoo5vTx2bjHaTkE9ozbo25h7eU4SuNUSAGmNnw2rD2MPxm7e5pzCxSgpyJfuzc7QwHbsNBe8vfqnEjhGW6/<0;1>/*
    ), 
    andor(
        multi(1,
            [362a0411/48h/0h/2h/2h]xpub6EynZEts5RmopiuAcaiF1vWnGHmNxvgrqmvvXGtyCJVRdZGP7bpRcGt7xVQUfWRQAQk8wNYdnMg5bxr38HkGYMQCkVxorUMCvphZy5HgPwU/<0;1>/*,
            [dbd96ebc/48h/0h/2h/2h]xpub6EgSRG4jge1K5rdMQFe24VSHXSYzjzFF7FvPWAkz7vWs4E7abo1i34AcyZsHhXwvpDzjEv2MqTMXBcATVfH2En3YJfpvgpE25JHRxF87Sus/<0;1>/*,
            [feda4d11/48h/0h/2h/2h]xpub6DkehEoVVoixuc9aqWdPvUAFy9zWbdigqcF6DPQThESgkJbU1TCS2iSXjB7BP2ToGgU5yErX2Ev1fp88b81NW3UGzyHdV3AsFGw7s2KEQRi/<0;1>/*),
            older(4255897),
    and_v(v:multi(2,
            [362a0411/48h/0h/1h/2h]xpub6DnFr7Ff8NbjcdrN2awC164Dgz64JHFCYF6YK18fqVqjZFUwQPQRyz38LR4C6QzLkHfbnttUPqWwL4FEvoA4uzS86YssWN8emQZjXiyuPtz/<0;1>/*,
            [dbd96ebc/48h/0h/1h/2h]xpub6EyjEwQ2rzHYttiTJUPW8sJxpnQhGDJCnBjK73ZGChK512fKfcdGQWhPoagrYhdJR8N1NcPKXwaTyh8tTLkj7Ej3bsZpYBNFiB2ZRQ8NyDa/<0;1>/*,
            [feda4d11/48h/0h/1h/2h]xpub6EmMNcU5bZKw6Xq1wbjTwhnVkG14vjhA9mvjii2Mk3QW8byAYahqJm7qYJaT2gAANviWgSu7xs7n2rn4ndnZGbhnDWbZdT1V2PuU6uaDyMe/<0;1>/*),
            after(1735689600)))))
```

PSBT is
```
cHNidP8BAFICAAAAASXBTaCNPS6qvR5GCFdjXDS5axe2fUfqsvpGV/ZGS5G7AAAAAAD9////Ac0BAAAAAAAAFgAUYLlCcpDeQDVWvogTYvXdl9WrLevmNAsATwEEiLIeBB3mM8yAAAACoI52PAvM15x1l2UN9/FXJ1IKkM8sTrhIu2uJMGDKJmwDE3k+Yo8GntAiSwsBlAqyBludB7E+uiinHVK8IK7yRWUU/tpNETAAAIAAAACAAgAAgAIAAIBPAQSIsh4EIaqojYAAAALk8s98JdgslIjNHNg8Q8LUKnEGVaOsMYq+wMpABV+IlANCRGGG/oLgrr1xGKAIWJlRlPRXXHctYlfSlNVIu+Od5xQ2KgQRMAAAgAAAAIABAACAAgAAgE8BBIiyHgQh3uopgAAAApYWSvGyPwA478hFMZASbs4ZfN3HX8PkxsLzBD6QaJstAlLRPfVyqA0UaidITf+u7AN7hmPc4y1X7KpUDNeUQs9lFDYqBBEwAACAAAAAgAAAAIACAACATwEEiLIeBJu55nKAAAACgCw2Cr9w/jZPXqziUxOUJXcG4Hok1qov9oILsTfgVe0CwJDpmbcmGLgmbZppKOR3wq9XR8o6QpristcopneM37MU/tpNETAAAIAAAACAAAAAgAIAAIBPAQSIsh4EnBItW4AAAAJdVTZLqFVQZT1XoBg6uPe4iiZM2vO8asAQTLBkY2/ouwN0HIvYJbHLPy2agVHcc1o3W1R1DxKohzKR3fWf4hCMMRTb2W68MAAAgAAAAIACAACAAgAAgE8BBIiyHgSnmHGGgAAAAkIScKz40q7/ldPXo8XTk7kIJ9Zbyuacsyct4Hd5tyjeAuJkwomsZNtHWOnOXqmjhG7I/b5qFB4EkEsDdqJru9FoFP7aTREwAACAAAAAgAEAAIACAACATwEEiLIeBMSg3YKAAAAC+pqzLMN+sCRIyUSy6joi+vAmcNT79y7TWQ+f9TGt9a8DGZZoE4Q2wT7if6PAXroDLhQX3uLnj4UUsQH6qlhRPBsU29luvDAAAIAAAACAAQAAgAIAAIBPAQSIsh4ExMMxU4AAAAKRSY8cmwKapqXPRnfaeUV7W9OGCzUBkTOeDGdahxaZYgPJY4Fps53xfoH8ZBIdGJmrhwSDiTn9hwyioVpcW6FKnhQ2KgQRMAAAgAAAAIACAACAAgAAgE8BBIiyHgTYsCYSgAAAAnCwHzlI8niTyPXJSkHhJ8tY8v95FLCbGFRCpnWsUyQTApjw5Us+xgoFIwrwFlHZi0J5sAjBEsHs3QKEeqY9h2LwFNvZbrwwAACAAAAAgAAAAIACAACAEPwJTXlDaXRhZGVsADYqBBEGYml0Ym94EPwJTXlDaXRhZGVsANvZbrwHbGVkZ2VycxD8CU15Q2l0YWRlbAD+2k0RB2xlZGdlcngAAQBeAgAAAAHbNEY1Pt0NXS2iVuHx1e1LQAEyKwr0/6B9kOyHrlMIAwAAAAAA/f///wFMAwAAAAAAACIAIMqifl87hvgRLE9CZObEFnU6ivXsg3CsyQLlblDVaKpy5jQLAAEBK0wDAAAAAAAAIgAgyqJ+XzuG+BEsT0Jk5sQWdTqK9eyDcKzJAuVuUNVoqnIBAwQBAAAAAQX9TAFjUyECe71bmLmw7veVKqQIGsIfOc+8osIePjeqnefJIv2URochA8FAk81unqFF2NZ4i/LvZqansSFSQCWzMUgM1/Pd7ifSIQIVl1AlMNr7+xEoGf0dgznJhdJV14l8S3ptvgvbf3+e5lOuZ1EhAputjaPGw3LMNNYt96mb7gyS00KaVTK3woGpoNkC2enfIQPtLK6QVtwsjQeImna88nk6WfgBpPreFhmn3ZuFYe8DmiEDUCFulwOwZ+YE+PpZ4idGXOAH9jL/+SRqTwGspRbIWttTrmRSIQMralk4UO+D7915SCZZgA2GCMSiNf23JRdUPfwP+GBDbiEDj+cYks0De3c9rsVKWrhwfP+tJVhhPrq3eR6GyfHrcXIhAx7ERqswS9MXz2iIHeFw59foRkY3Vu+KI+rUaky184rzU68EgIV0Z7FnA5nwQLJoaCIGAhWXUCUw2vv7ESgZ/R2DOcmF0lXXiXxLem2+C9t/f57mHP7aTREwAACAAAAAgAAAAIACAACAAAAAAAAAAAAiBgJ7vVuYubDu95UqpAgawh85z7yiwh4+N6qd58ki/ZRGhxw2KgQRMAAAgAAAAIAAAACAAgAAgAAAAAAAAAAAIgYCm62No8bDcsw01i33qZvuDJLTQppVMrfCgamg2QLZ6d8cNioEETAAAIAAAACAAgAAgAIAAIAAAAAAAAAAACIGAx7ERqswS9MXz2iIHeFw59foRkY3Vu+KI+rUaky184rzHP7aTREwAACAAAAAgAEAAIACAACAAAAAAAAAAAAiBgMralk4UO+D7915SCZZgA2GCMSiNf23JRdUPfwP+GBDbhw2KgQRMAAAgAAAAIABAACAAgAAgAAAAAAAAAAAIgYDUCFulwOwZ+YE+PpZ4idGXOAH9jL/+SRqTwGspRbIWtsc/tpNETAAAIAAAACAAgAAgAIAAIAAAAAAAAAAACIGA4/nGJLNA3t3Pa7FSlq4cHz/rSVYYT66t3kehsnx63FyHNvZbrwwAACAAAAAgAEAAIACAACAAAAAAAAAAAAiBgPBQJPNbp6hRdjWeIvy72amp7EhUkAlszFIDNfz3e4n0hzb2W68MAAAgAAAAIAAAACAAgAAgAAAAAAAAAAAIgYD7SyukFbcLI0HiJp2vPJ5Oln4AaT63hYZp92bhWHvA5oc29luvDAAAIAAAACAAgAAgAIAAIAAAAAAAAAAAAAA
```

Txid which is being spent: [`bb914b46f65746fab2ea477db6176bb9345c635708461ebdaa2e3d8da04dc125:0`](https://mempool.space/tx/bb914b46f65746fab2ea477db6176bb9345c635708461ebdaa2e3d8da04dc125)

Yes, that's mainnet. I do write code and debug it only on mainnet, not testnet. Because that is the way.

And yes, I know, that is a crazy descriptor. Here is what it means:
[image]
[image]

benma, 2023-09-08:
As mentioned in https://github.com/bitcoin-core/HWI/issues/706#issuecomment-1710194139, HWI does not support miniscript policies for the BitBox02, as the policy has to be sent to the BitBox02, and it is not part of PSBT (yet?) and HWI does not currently provide the API to send it separately. The policy that needs to be sent to the BitBox02 is following [this BIP](https://github.com/bigspider/bips/blob/bip-wallet-policies/bip-wallet-policies.mediawiki).

The [Rust BitBox02 library](https://github.com/digitalbitbox/bitbox-api-rs) on the other hand, which is being integrated into Rust async-hwi, does support miniscript policies already and could be a better choice for you.

To get support into HWI, we should open an issue there and discuss with the maintainers what a good way would be. Ideally the policy would be encoded in the PSBT itself, but afaik there is no standard around that yet.

dr-orlovsky, 2023-09-08:
Thank you for all the details. I will do an issue in HWI repo as you have suggested.

However, one more thing: it is not clear for me why we need to send the policy to the BitBox if the policy can be re-constructed using miniscript library from the `PSBT_IN_WITNESS_SCRIPT` (for pre-taproot) and `PSBT_IN_TAP_LEAF_SCRIPT` (for taproot), which is already present in the PSBT - and together with key derivation fields (like `PSBT_GLOBAL_XPUB`, `PSBT_IN_BIP32_DERIVATION`, `PSBT_IN_TAP_BIP32_DERIVATION`, `PSBT_IN_TAP_INTERNAL_KEY`) the wallet ALREADY has all information to register the policy.

So my point is that **no additional** PSBT fields or call to API to send it separately should be required. The algorithm is the following:
1. Check which inputs have keys (from BIP32 derivation fields) derived from the device seed, for each of them:
2. If the input is pre-taproot, reconstruct miniscript policy from `PSBT_IN_WITNESS_SCRIPT` and register it;
3. If the input is taproot, reconstruct TapTree (again using miniscript library) from the individual control blocks and register it;
3.a. If PSBT doesn't contain `PSBT_IN_TAP_LEAF_SCRIPT` to reconstruct the full tree, fail with error message.

dr-orlovsky, 2023-09-08:
Ok, I have read [all the arguments in the discussion(https://github.com/bitcoin/bips/pull/1389) and see why you need a new "descriptor template" language (next to miniscript, miniscript policy, output descriptor and xkey descriptor languages).

However, I still doubt whether a separate API call or a custom PSBT field is required, since all information to build "descriptor template" and display it to the user in the device UI is already there in PSBT (all my arguments/algorithm from the comment above  still apply).

dr-orlovsky, 2023-09-08:
> To get support into HWI, we should open an issue there and discuss with the maintainers what a good way would be. 

It also seems there is already a WIP implementation for this: https://github.com/bitcoin-core/HWI/pull/647

benma, 2023-09-09:
Isn't PSBT_IN_TAP_LEAF_SCRIPT only part of the tree, but one needs the full tree to reconstruct the full policy?

The BitBox02 also needs the policy to register the wallet on the device before any signing or receiving, and is also needed to display receive addresses, so it must be available on the host anyway.

dr-orlovsky, 2023-09-09:
Each individual `PSBT_IN_TAP_LEAF_SCRIPT` represents a single taptree leaf - but PSBT can provide multiple - up to covering all leafs (property is key-mapped, where the key is control block, so you can have many of the keys for the same property). In such way the wallet communicates whole taptree - and if some leaf is missed, that device may notify the user and reject PSBT.

Thus, you do not need custom PSBT structures. 

Next, you use tap tree to reconstruct miniscript descriptor - and from it build your policy template and register it.

benma, 2023-09-09:
I prefer the explicit approach the BitBox02 took already. The code complexity and binary bloat to support reconstructing a policy string is significant, including all the multipath `<n;m>` that could appear. The compatibility issues would be greater, as PSBTs will not normally contain all leafs. 

Furthermore, I am not sure the reconstructed policy will always be unique, and may not match the exact way it is shown in a coordinator wallet (e.g. `@1/**` vs `@1/<0;1>/*`, etcc.). There was also talk about breaking miniscript roundtrips by extending `and(X1,X1)` to `and(X1,X2,X3,...)`, which would be equivalent to `and(...(and(X1,X2),X3),...)` - not sure if that was abandoned or not.

> Next, you use tap tree to reconstruct miniscript descriptor - and from it build your policy template and register it.

That is a UX problem as well, if you want to e.g. show in the coordinator wallet that the wallet should be registered, and give it a name, the registration must be separate. 

For now the BitBox02 will stick to explicit registration and sending the policy for receive addresses and signing.

closed 2023-09-09 by benma

dr-orlovsky, 2023-09-10:
I am not sure we are talking about the same thing here.

My assumption is that you are already using `rust-miniscript` in your firmware, thus everything I have described is available out of the box already inside your binary - you just need to call one more method. At the same time asking to add PSBT field will end up in "standardization hell" and will take forever.

If you are not using `rust-miniscript`, everything you said makes perfect sense.

benma, 2023-09-10:
We are using rust-miniscript, but assembling a policy string including multipaths is not part of it. Not sure if even producing the descriptor from the leaf scripts is there or not. Regardless, just because some of the code already exists it does not mean it is not complex or that it does not induce significant bloat. This work is best offloaded to host coordinators, which usually already have the policy. 

There are more reasons to prefer the explicit approach. Some of them given above (UX, uniqueness, etc.), but also the BitBox02 streams the transaction but displays which account/policy it spends from in the beginning before any inputs are streamed. If the info is reconstructed from inputs, the signing/streaming API of the BitBox02 would have to be changed significantly.

> At the same time asking to add a PSBT field will end up in "standardization hell" and will take forever.

One can send it separately of the PSBT (Liana is doing this for the Ledger and the BitBox02) and how the [HWI PR](https://github.com/bitcoin-core/HWI/pull/647) you linked does it, or BIP174 Proprietary keys could be made use of until a standard emerges.