BitBoxSwiss/bitbox02-firmware #1147: Error "Previous transaction missing"

swiss-chris, 2023-11-07:
[image]

I tried signing a PSBT of a 2-of-2 multisig. After confirming the multisig wallet on the BitBox02, when the device said "Loading transaction..." Sparrow gives me the above error "Previous transaction missing" (see screenshot).

Signing the exact same PSBT with ColdCard (using the same seed phrase) works.

Can you confirm that this is most likely a bug in the BitBox02 ? What can I do to fix it ?

benma, 2023-11-09:
It is not a bug - for Native Segwit inputs, the full previous transaction is required so that the BitBox02 can safely calculate and display the transaction fee. 

How did you generate the PSBT? I thought that Sparrow included the previous transactions, as BitBox02 generally seems to work for multisig with Sparrow.

swiss-chris, 2023-11-09:
I created the transaction directly from the Sparrow wallet containing the 2-of-2 Native Segwit Multisig.

benma, 2023-11-09:
@craigraw any idea? At least BitBox02 single-sig works as far as I know, which also requires the previous transactions. I thought multisig worked as well (it seems you tested that recently - https://github.com/bitcoin-core/HWI/issues/700).

craigraw, 2023-11-09:
Sparrow does provide the full previous transaction (technically, the `PSBT_IN_NON_WITNESS_UTXO` field) for each input in the PSBT where it knows a Bitbox02 is signing. I've just retested this, and signing a Native Segwit multisig transaction worked fine.

Is the type of the keystore in the Settings tab a Bitbox 02 Connected Wallet?

[image]

swiss-chris, 2023-11-15:
It was originally set to Type "Coldcard" when I created this ticket.

However, today I re-imported the same seed (same master fingerprint, derivation path, xpub etc.) from my Bitbox wallet, created a new transaction, and when trying to sign with Bitbox, I still get the exact same error message.

benma, 2023-11-16:
@swiss-chris maybe it's better to open this issue at Sparrow - I don't think there is an issue in the BitBox02 firmware or libraries. Unless @craigraw wants to continue here :smile:

benma, 2023-11-16:
That being said, if you are stuck and need to make a transaction, I suggest trying it with Electrum, which also has support for the BitBox02 and multisig.

craigraw, 2023-11-16:
> It was originally set to Type "Coldcard" when I created this ticket.

This was almost certainly the cause then. Coldcard does not require the full previous transaction for every segwit UTXO.

> However, today I re-imported the same seed (same master fingerprint, derivation path, xpub etc.) from my Bitbox wallet, created a new transaction, and when trying to sign with Bitbox, I still get the exact same error message.

Did you save the wallet afterwards (click Apply)? You could also try recreating as a new wallet.

>  I don't think there is an issue in the BitBox02 firmware or libraries

@benma Respectfully, I think the decision to require the full previous transaction for every segwit UTXO to mitigate an unlikely attack vector was a mistake. It bloats the PSBT to a size that creates many downstream problems - for example parsing on certain limited RAM devices and transmitting over QR. Ultimately, this means the coordinator needs to know what kinds of signers are present, leading to issues like this. Other mitigations (for example the one used by Coldcard) have proved less invasive.

swiss-chris, 2023-11-17:
@craigraw Thanks for your response. I can confirm that after replacing the hardware wallet in the sparrow wallet settings and then saving the sparrow wallet, I can indeed sign with my BitBox02.

However, I also noticed that if I backup my sparrow wallet in BSMS format and then import that as a new sparrow wallet, I can sign the transaction with my BitBox02 even though the signing device / xpub is marked as "watch-only" wallet. This makes me think it would be more convenient for my daily use if I always work with watch-only wallets instead of setting up the Sparrow wallet to expect vendor-specific configurations.

But then I am reminded of my previous ticket https://github.com/sparrowwallet/sparrow/issues/1067 where I realized I could not sign messages unless I imported the specific hardware wallet first. 

So now I am stuck. If I leave it as watch-only I cannot sign messages with my BitBox02, and if I import the Coldcard wallet as vendor-specific signer I cannot sign with the BitBox02. I assume I was experiencing other issues that made me import the Coldcard wallet, although I am not sure. Maybe I could import the BitBox02 and still be able to sign transactions and messages with the ColdCard if needed? All in all, for my own use it's confusing, and imagining someone trying to access my Bitcoin after my death makes me even less certain that they will manage.

If you would like me to post this as a new question under https://github.com/sparrowwallet/sparrow/issues/new I would be happy to.

benma, 2023-11-18:
> @benma Respectfully, I think the decision to require the full previous transaction for every segwit UTXO to mitigate an unlikely attack vector was a mistake. It bloats the PSBT to a size that [...]

I of course see the downsides, but most attack vectors are unlikely to be exploited as there are often lower hanging fruits for attackers (like phishing). That does not mean we should leave them unpatched :thinking: I am not sure what Coldcard does, but it is unlikely to be a complete fix unless they stream the previous transactions and verify the input amounts that way. As a remote attack that can put all funds in the account at risk, it is important to mitigate it.

We could consider converting this from an error to a warning, but I am not a big fan of this, and the message is basically equivalent to "continue if you trust your computer is not compromised".

Fyi, the same issue does not exist in Taproot - the problem should disappear on its own with Taproot adoption. When minitapscript? :smile:

craigraw, 2023-11-20:
> Maybe I could import the BitBox02 and still be able to sign transactions and messages with the ColdCard if needed?

I would recommend this route. The mitigation we are discussing only adds information to the PSBT. In some situations this can cause very large PSBTs that older models of the Coldcard may struggle with if you are using them as a signing device, but in general it should be ok. So long as the master fingerprints on the Coldcard and the BitBox02 match, there should be no problems signing messages or transactions.

I hear you on the complexity, and I'll consider making the message signing work the same as signing a transaction wrt popping up the devices dialog on a watch only wallet. Fwiw, as you can see I find the mitigation frustrating to deal with as well.

craigraw, 2023-11-20:
> I am not sure what Coldcard does, but it is unlikely to be a complete fix

@benma Here is a description of their approach: https://blog.coinkite.com/version-3.1.4-released/ - is such an approach possible with the BitBox02?

> Fyi, the same issue does not exist in Taproot - the problem should disappear on its own with Taproot adoption. When minitapscript?

Indeed - but I think FROST is probably the best Taproot multisig option in the works. Hoping we get standards there soon.

swiss-chris, 2023-12-28:
Please delete this issue. Thanks 🙏